OpenHMD/OpenHMD #276: Samsung Odyssey Plus screen stays dark

johnr14, 2020-01-17:
I am trying hard to get SteamVR working with openhmd without success. Backtracking, I compiled the opengl example and I have a black screen. So I checked more closely the output of simple and I think the problem comes from the 0 x 0 screen size !

I am using the dev-wmr-controllers build, but got the same error with master. Is there something I am missing ? I added the udev and rebooted. It's as if the screen is in sleep mode and putting headset on head doesn't change a thing. Was there an update to the device firmware lately that changed something like #174 had ?Also #272 seems to have reported it ? 

Q: Is there a special permission to set for access to screen HDMI-A-0 ?

Are there other running O+ without glitch ? The audio +/- controls works and change the main mixer sound.

I am available to try debugging this.
Thanks

System : 
Fedora 31 (up to date) 
kernel 5.4.8-200.fc31.x86_64
KDE 5.17.5
OpenGL version string: 4.5 (Compatibility Profile) Mesa 19.2.8
AMD 1900X
AMD Vega64
Samsung Odyssey+

This is the output ou simple : 
```
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
[... 198 lines cut ...]
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 74 
                supported: 74
        non-desktop: 1 
                range: (0, 1)
```

johnr14, 2020-01-17:
I rebased on f41816db1a8cea3c1c0dcee87f17ed4fda442a16 and it fix the size 0 x 0 ! But screen is still black. I didn't specify, but current setup works fine on W10 in case some one raised the bad cable/connector/etc issue.

simple: 
```
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 045e:0659 0008:0104:0000:0000:0104:02
[II] Opening
[... 140 lines cut ...]
[II] Read 10219-byte config data
[II] Model name: Samsung Windows Mixed Reality 800ZBA
[EE] {"CalibrationInformation":{"Cameras":[{"Intrinsics":{"ModelParameterCount":15,"ModelParameters":[0.50986164808273315,0.48059824109077454,0.42008697986602783,0.56038570404052734,0.79907500743865967,0.67329293489456177,0.0038568372838199139,1.0711227655410767,0.80461454391479492,0.087160795927047729,0.0029966279398649931,0.0010741509031504393,7.1934977313503623E-5,1.0643623681971803E-5,2.792189359664917],"ModelType":"CALIBRATION_LensDistortionModelRational6KT"},"Location":"CALIBRATION_CameraLocationHT0","Purpose":"CALIBRATION_CameraPurposeHeadTracking","MetricRadius":2.792189359664917,"Rt":{"Rotation":[1,0,0,0,1,0,0,0,1],"Translation":[0,0,0]},"SensorHeight":480,"SensorWidth":640,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}},{"Intrinsics":{"ModelParameterCount":15,"ModelParameters":[0.50890707969665527,0.4991055428981781,0.41541168093681335,0.55406659841537476,0.50283777713775635,0.2812899649143219,0.0060616401024162769,0.775051474571228,0.31480833888053894,0.049422327429056168,0.0065727368928492069,0.0066355052404105663,0.00017096112424042076,3.507217115839012E-5,2.7875354290008545],"ModelType":"CALIBRATION_LensDistortionModelRational6KT"},"Location":"CALIBRATION_CameraLocationHT1","Purpose":"CALIBRATION_CameraPurposeHeadTracking","MetricRadius":2.7875354290008545,"Rt":{"Rotation":[0.69421511888504028,-0.0052430024370551109,-0.71974849700927734,0.0085603352636098862,0.99996286630630493,0.00097243225900456309,0.71971666812896729,-0.0068363654427230358,0.69423425197601318],"Translation":[-0.09655163437128067,-0.00072534364880993962,-0.041091181337833405]},"SensorHeight":480,"SensorWidth":640,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}},{"Intrinsics":{"ModelParameterCount":15,"ModelParameters":[0.50568294525146484,0.48980224132537842,0.43511775135993958,0.52002924680709839,6.0497612953186035,-10.048660278320312,22.599639892578125,6.1337661743164063,-10.050261497497559,22.486785888671875,-0.0077081732451915741,0.0020905046258121729,0.00099883740767836571,-4.3223499233135954E-6,1.3999999761581421],"ModelType":"CALIBRATION_LensDistortionModelRational6KT"},"Location":"CALIBRATION_CameraLocationDO0","Purpose":"CALIBRATION_CameraPurposeDisplayObserver","MetricRadius":1.3999999761581421,"Rt":{"Rotation":[1,0,0,0,1,0,0,0,1],"Translation":[0,0,0]},"SensorHeight":2048,"SensorWidth":2448,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}},{"Intrinsics":{"ModelParameterCount":15,"ModelParameters":[0.50495320558547974,0.49421533942222595,0.43522077798843384,0.519997775554657,-0.29937207698822021,1.635257363319397,2.0507051944732666,-0.25487691164016724,1.5702483654022217,2.0532345771789551,-0.0011014774208888412,-0.011134925298392773,0.00073413283098489046,0.00052863958990201354,1.3999999761581421],"ModelType":"CALIBRATION_LensDistortionModelRational6KT"},"Location":"CALIBRATION_CameraLocationDO1","Purpose":"CALIBRATION_CameraPurposeDisplayObserver","MetricRadius":1.3999999761581421,"Rt":{"Rotation":[0.99999034404754639,-0.0036428447347134352,-0.0024573411792516708,0.0036511879879981279,0.99998754262924194,0.0033993404358625412,0.0024449273478239775,-0.0034082799684256315,0.99999117851257324],"Translation":[-0.065898001194000244,-3.4150158171541989E-5,-8.4063372923992574E-5]},"SensorHeight":2048,"SensorWidth":2448,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}}],"Displays":[{"Affine":[908.44085693359375,-0.5635298490524292,784.12884521484375,-0,907.9769287109375,798.8868408203125,0,0,1],"AssignedEye":"CALIBRATION_DisplayEyeLeft","DisplayHeight":1600,"DisplayWidth":2880,"Distort [... cut ...]
OpenGL Renderer: Radeon RX Vega (VEGA10, DRM 3.35.0, 5.4.8-200.fc31.x86_64, LLVM 9.0.0)
OpenGL Vendor: X.Org
OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.2.8
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
```

TheOnlyJoey, 2020-01-17:
For getting your display to output, it might be wise to read documentation https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode though your resolution not working looks like a configuration issue.
Might want to patch in that if configuration does not work, it should quit and print a message.

I don't have this device myself, my Acer WMR still works on latest firmware so its hard for me to debug at this moment.

johnr14, 2020-01-17:
I already ran xrandr and it did not change the disconnect status. I remember seeing a kernel param for amd and vr, might try to find the page again. But it may disable hdmi sound if I remember well.
Thanks

```
xrandr --output HDMI-A-0 --set non-desktop 0
```

```
HDMI-A-0 disconnected (normal left inverted right x axis y axis)
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 74 
                supported: 74
        non-desktop: 0 
                range: (0, 1)
```

johnr14, 2020-01-17:
There seems to be some issues on SteamVR with similar black screen. [#189](https://github.com/ValveSoftware/SteamVR-for-Linux/issues/189) and [#274](https://github.com/ValveSoftware/SteamVR-for-Linux/issues/274).

Also, just found #274 and it is where I saw the fix for the kernel, will try it when I have the time.
How is it possible to set  display mode to RGB 4:4:4 ? Does this mean linux VR on AMD is borked for the moment related to driver issue ?

I have a DK2 in storage, will try to get it out and try it out during the week.

I get the 3D display window on my main monitor currently and O+ is black on standby. I can't get xrandr HDMI-A-0 do display connected.

Is there a way to get verbose log of Xorg to check for errors accessing hmd ?

Thanks

johnr14, 2020-01-19:
Rebooted with `amdgpu.dc=0` and it's not working.

This is what is logged in journalctl when running the opengl demo : 
```
Jan 18 19:51:35 wksfed kwin_x11[2271]: qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 16090, resource id: 98566164, major code: 15 (QueryTree), minor code: 0
Jan 18 19:51:35 wksfed kwin_x11[2271]: qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 16095, resource id: 98566164, major code: 18 (ChangeProperty), minor code: 0
```

There are no kernel events when plugging/unplugging the hdmi cable to the computer.

Could the device be in sleep mode and require a usb wake up call. 
**The O+ do have a front sensor to check if the display is put on the head ?**

Investigating that, I went to enable usbdebug following[ kernel doc](https://www.kernel.org/doc/Documentation/usb/usbmon.txt). 

On reading I get the following when hiding the front face facing captor : 
```
ffff9bd35b376b40 2249249280 C Ii:7:008:1 0:128 4 = 10014606
ffff9bd35b376b40 2249249293 S Ii:7:008:1 -115:128 64 <
```
And this when I unhide it.
```
ffff9bd35b376b40 2250273281 C Ii:7:008:1 0:128 4 = 10004706
ffff9bd35b376b40 2250273295 S Ii:7:008:1 -115:128 64 <
```
When the opengl demo starts, I don't get any event on that interface ?
Will try to see if windows sends events to wake up the hmd on that port... Not sure if I will succeed or if I am going the right way.

To be continued...

johnr14, 2020-01-19:
Ok, got USBPcap installed and captured packets !

Found packet matching the detection of the sensor :
1b00506084530ae6ffff0000000009000104000600810104000000**10014706**

After 1.176 sec, we start seeing bulk transfers of data from headset to pc. Those are probably the images from the front camera to get tracking working. By that point, the hmd should be displaying images.

I'll put the logged packets here as I am not sure I can debug this myself. I used wireshark to open the file, may need to rename it to **.pcap**.

[testpackets.log](https://github.com/OpenHMD/OpenHMD/files/4082223/testpackets.log)

johnr14, 2020-01-19:
Getting around the USB packets... Hope this can help !

I made a longer log, from press to unpress the head sensor once the screen lit up.
[startingshort.log](https://github.com/OpenHMD/OpenHMD/files/4082281/startingshort.log)

I will list suspects packets : 
57	0.078823	host	4.5.0	USB	27	Unknown type 7f
`1b00a0e8b5640ae6ffffffffffff3d00000400050000ff00000000`
287	0.397392	host	4.5.0	USB	27	Unknown type 7f
`1b0020e835660ae6ffffffffffff3d00000400050000ff00000000`
// seems to get audio ready to receive data
451	0.623445	host	4.5.4	USB	27	URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL
`1b00a018fd600ae6ffff000000001e00000400050004fe00000000`
// could it be to ready the display for hdmi data ?
541	0.709690	host	4.5.1	USB	27	URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL
`1b00a0286b670ae6ffff000000001e00000400050081fe00000000`
// and this one
1868	1.880731	host	4.6.1	USB	27	URB_INTERRUPT in
`1b00e05683530ae6ffff0000000009000004000600810100000000`

I made a second log of when I close the WMR window 
[close_smaller.log](https://github.com/OpenHMD/OpenHMD/files/4082278/close_smaller.log)
// Audio ?
1	0.000000	host	4.5.4	USB	27	URB_FUNCTION_ABORT_PIPE
`1b00a0383c680ae6ffff000000000200000400050004fe00000000`

// Video ?
438	0.566762	host	4.6.1	USB	27	URB_INTERRUPT in
`1b00506084530ae6ffff0000000009000004000600810100000000`
// Last to 4.6.1
466	0.598760	host	4.6.1	USB	27	URB_INTERRUPT in
`1b00e05683530ae6ffff0000000009000004000600810100000000`

Mangmasta, 2020-01-19:
I have the Odyssey+ with an AMD card and I can confirm that I am experiencing this exact same issue as well.  This is the first time I tried setting up OpenHMD on linux so I am unsure if it was an update that broke things or what.  I notice that the screen does turn on in windows when picking up the unit and goes to sleep when windows mixed reality program is closed.  I have not installed wireshark to capture packets seeing as you have already done that.  Please let me know though if you need any debug info from me.

johnr14, 2020-01-19:
@Mangmasta thanks, was wondering if I was alone in this situation. Still, it would be useful if someone with a nvidia card can post if it works for him.

**Odyssey+ sale pitch**
I think the Odyssey+ is a realy good headset and is over the top for performance/price ratio. There are on sale at 289$USD and sometime lower. They are a bang for the buck! And visuals are good.
Only thing missing is linux support ! 

The way I see it, once we get it to work, is to get the camera stream and process it in opencv.
First is to get positional data using something like [SVO](https://hackaday.com/2016/04/02/insanely-quick-3d-tracking-with-1-camera/). 
After that, get hand tracking like [this](https://www.reddit.com/r/ValveIndex/comments/dckot2/quick_tensorflow_hand_tracking_test_with_valve/) with [mediapipe](https://ai.googleblog.com/2019/08/on-device-real-time-hand-tracking-with.html). 
Then might mix ps move for [gun](https://www.youtube.com/watch?v=XiW6avvN8q4) and wondering if positionnal tracking could be done with onboard camera
Finaly Kinect v2 for [body movement like driver4vr](https://www.youtube.com/watch?v=RUuWHARX7-M)/[ps move controllers](https://www.youtube.com/watch?v=diPGvTCJtyw).

Then those WMR would be cost effective great VR gear. I personnaly think that MS should just opensource WMR to keep it alive in the fierce market between FB/VALVE/SONY.
**End Odyssey+ sale pitch**

Back on track, I tried to write a python script to send a packet to the device, for now not much luck as code doesn't work. And I'm more of a linux enthusiast than a full fledged programmer.

<details>
<summary>WIP python code</summary>

```
import usb.core
import usb.util
from pprint import pprint

""" DEBUG
busses = usb.busses()
for bus in busses:
    devices = bus.devices
    for dev in devices:
        pprint(dev)
        for config in dev.configurations:
            pprint(config)
            for intf in config.interfaces:
                print("Interface "+str(intf[0].interfaceNumber))
                for alt in intf:
                    print("Alt setting "+str(alt.alternateSetting))
                    for ep in alt.endpoints:
                        print("endpoint "+str(ep.address))
"""

def odyssey_write(command):
    VENDOR_ID=0x045e
    PRODUCT_ID=0x0659
    INTERFACE_NB=0
    SETTING_NB=0 # always 0?
    ENDPOINT_NBq=0
    
    device = usb.core.find(idVendor=VENDOR_ID, idProduct=PRODUCT_ID)
    if device is None:
        print("Is the headset on?")
        sys.exit(1)
    else:
        print("Got device !")
        pprint(device)
        device.reset()
    
#    if device.is_kernel_driver_active(CONFIGURATION_NB):
#        print("Detaching kernel driver")
#        device.detach_kernel_driver(CONFIGURATION_NB)
    
    configuration = device.get_active_configuration()
    print("Got configuration:")
    # there are 3 configuration #0,1,2
    #print(configuration)
    
    
    print("DEBUG: LISTING ALL ENDPOINTS FOR INTERFACES")
    for INTERFACE_NB in range(3):
        interface = configuration[(INTERFACE_NB, SETTING_NB)]
        print("Got interface :")
        pprint(interface)
        
        for ENDPOINT_NB in range(2):
            endpoint = interface[ENDPOINT_NB]
            print("Got endpoint :")
            print(endpoint)
    
    """ BAD ???
    INTERFACE_NB=1
    ENDPOINT_NB=0
    interface = configuration[(INTERFACE_NB, SETTING_NB)]
    endpoint = interface[ENDPOINT_NB]
    print("Endpoint: ")
    print(endpoint)
    """
    
    #https://github.com/walac/pyusb/blob/master/docs/tutorial.rst
    
    ##data = device.crtl_transfer(0b10100001, 0x01, 3<<8, 0, 4)
    ##print(data)
    
    
    try :
        endpoint.write(command)
    except USBError:
        print("Error")
    
    
    print("Wrote cmd to endpoint")
    usb.util.dispose_resources(device)

startcmd = '\x1b\00\xa0\x28\x6b\x67\x0a\xe6\xff\xff\x00\x00\x00\x00\x1e\x00\x00\x04\x00\x05\x00\x81\xfe\x00\x00\x00\x00'
#startcmd2 = ''
closecmd = ''

odyssey_write(startcmd)
```

</details>

Mangmasta, 2020-01-20:
OK, I actually wanted to try capturing packets as well.  I am a noob to it but I read a tutorial and I THINK I understand what I'm doing.  The packets that are being sent/received for me are completely different from yours @johnr14 .

I started capturing then started the WMR portal, let the screens turn on, I picked it up, moved it around a bit and then shut down the portal app.  I did this multiple times each time it gave me pretty much the same packets.  I attached one set of them here.

Very strange though, if Im reading this right when I opened up the WMR portal, the first thing that happened was a packet FROM the headset?  This capture is complete.  No events happened before or after the portal app was closed.  I would think to wake up the screens the first thing that would happen would be a packet from the computer to the headset...  Could it be possible that the data pins on the HDMI port are used or something?  IDK im just throwing ideas out there.
[filteredHMD.zip](https://github.com/OpenHMD/OpenHMD/files/4083971/filteredHMD.zip)

johnr14, 2020-01-20:
@Mangmasta  Unfortunately, your file is mostly empty. Check filesize. I only see packates (10) going to 2.5.1.

There are packets for audio and video (from camera) that can be large... First let's check if you have the same device configuration by running this python script. There are a few USB devices and naming schemes are weird for me.

<details>
<summary>USB device info</summary>

```
import usb.core
import usb.util
from pprint import pprint

"""
busses = usb.busses()
for bus in busses:
    devices = bus.devices
    for dev in devices:
        pprint(dev)
        for config in dev.configurations:
            pprint(config)
            for intf in config.interfaces:
                print("Interface "+str(intf[0].interfaceNumber))
                for alt in intf:
                    print("Alt setting "+str(alt.alternateSetting))
                    for ep in alt.endpoints:
                        print("endpoint "+str(ep.address))
"""

def odyssey_info():
    VENDOR_ID=0x045e
    PRODUCT_ID=0x0659
    INTERFACE_NB=0
    SETTING_NB=0 # always 0?
    ENDPOINT_NBq=0
    
    device = usb.core.find(idVendor=VENDOR_ID, idProduct=PRODUCT_ID)
    if device is None:
        print("Is the headset on?")
        sys.exit(1)
    else:
        print("Got device !")
        pprint(device)
        print(device)
        device.reset()
       
    usb.util.dispose_resources(device)

odyssey_info()
```
</details>

<details>
<summary>My details</summary>

```
Got device !
<DEVICE ID 045e:0659 on Bus 008 Address 003>
DEVICE ID 045e:0659 on Bus 008 Address 003 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x300 USB 3.0
 bDeviceClass           :   0xef Miscellaneous
 bDeviceSubClass        :    0x2
 bDeviceProtocol        :    0x1
 bMaxPacketSize0        :    0x9 (9 bytes)
 idVendor               : 0x045e
 idProduct              : 0x0659
 bcdDevice              :    0x5 Device 0.05
 iManufacturer          :    0x1 Microsoft
 iProduct               :    0x2 HoloLens Sensors
[... 193 lines cut ...]
hsize:                   0.103812 
vsize:                   0.051905 
lens separation:         0.063000 
lens vcenter:            0.025953 
left eye fov:            1.658063 
right eye fov:           1.658063 
left eye aspect:         1.000000 
right eye aspect:        1.000000 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:      
```  
</details>

Following that, you can retry capturing packets. Be sure that you have only your headset plugged in a hub or you will log mouse/keyboard and any activity on that hub. See my previous image. Then run the capture program, select the # of root usb hub to monitor (1 in my case) press then write filename, press enter and a pop-up will ask permission. click yes. It will now log packets.

**First log the turning on of display :** 
Press with your finder the flashing light sensor inside the helmet. Try not to move it around as it will generate a bunch of IMU data packets. Keep your finger on the sensor, wait for display to lit up on hmd and remove finder. Close the capture window. Now, the sensor will serve to identify where to look for the usb communications that may turn on the display.

**Second the turning off of display**
Start headset and wait for display. Start logging. Close VR window ans wait for display to turn off. Stop logging. 

Using those 2 logs, we should be able to locate the cmd to turn on/ turn off hmd display. And comparing between us, we could narrow suspect packets.

**I am stuck at the python code for sending packets to hmd. If anyone got some more info / care to give help would be welcome.**

Mangmasta, 2020-01-20:
Attached is a txt file with all that logs stuff.  It seems like we have the same device.
[samsungOdysseyPlus.txt](https://github.com/OpenHMD/OpenHMD/files/4084165/samsungOdysseyPlus.txt)

I understand your way, I can give it a try but I tried capturing it a different way I believe is correct.  After installing USBpcap on windows I can see it my root usb buses in wireshark.  I just used wireshark to start capturing that bus then used the filter:
```
usb.src == "2.5.1"|| usb.dst == "2.5.1"
```
according to the USBpcapCMD.exe program 2.5.1 is the port which the headset is plugged into.  That log I put in the previous post is the result of me running wireshark with that filter.
I just launched the WMR portal on my computer (I did not touch or move the headset), looking from the side I can see the screens inside turn on, I picked up the headset and then put it back down, then I exited the WMR portal.

You're right though Im not sure why it does not show the packets for sensor data with my method...  I will try doing it your way.

Can you check to see if the packets that I found are at all found in your logs?

johnr14, 2020-01-20:
@Mangmasta I made a compilation of my findings [here](https://github.com/johnr14/openhmd-wmr-usb).

We are looking at different data, I am working with endpoint 0x05, the same that is used to enable video stream from the onboard camera, and your log has endpoint 0x81. But since I'm no expert, and your log is filtered, I am not able to make sense of it. You should be able to see the cmd to start the camera as mentioned [here](https://github.com/OpenHMD/OpenHMD/issues/200) if you are looking at the right thing. Thanks for your solidarity !

johnr14, 2020-01-21:
I got my **Oculus DK2 working on my setup** with AMD Vega64 ! 
I had to set amdgpu.dc=0 in kernel param. Will have to retry without.

**This means that the Odyssey + is missing something for it to work.** 
The usb devices (audio and camera) are fully functional and tested.

1) **a bug in amdgpu kernel** ? that prevents screen detection ? 
-> Try Pro driver (the not open amd driver) to see if it works
2) A **initialization cmd** must be sent to the hmd (usb or hdmi?)
-> Trying to log and find suspect usb packets and replay them.

<details>
<summary>xrandr --props for DK2</summary>

```
HDMI-A-0 disconnected (normal left inverted right x axis y axis)
        EDID: 
                00ffffffffffff003ed2030000000000
                0a18010380000078e260b1aa5540b623
                0c505400000001010101010101010101
                0101010101017240383a40800d70210a
                1600477e0000001a000000fc00526966
                7420444b320a20202020000000ff0048
                314445343134364b4843384b000000fd
                00384d1e9611000a2020202020200152
                020304038e33383a40800d70210a1600
                477e0000001ade3d383a40800d70210a
                1600477e0000001ad933383a40b41830
                210ac600477e0000001a000000000000
                00000000000000000000000000000000
                00000000000000000000000000000000
                00000000000000000000000000000000
                000000000000000000000000000000f0
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 74 
                supported: 74
        non-desktop: 1 
                range: (0, 1)
   1080x1920     75.00 +  72.00    60.00  
   1080x948     119.99  
   1024x768      75.00  
   800x600       75.00  
   640x480       75.00  
```
</details>
<details>
<summary>xrandr --props for Samsung Odyssey +</summary>

```
HDMI-A-0 disconnected (normal left inverted right x axis y axis)
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 74 
                supported: 74
        non-desktop: 0 
                range: (0, 1)
```
</details>

johnr14, 2020-01-21:
Playing around xrandr ! Still no success getting HDMI to show connected...

I ran : 
```
xrandr --output HDMI-A-0 --prop --newmode "2880x1600_90.00" 606.24 2880 3120 3440 4000 1600 1601 1604 1684 -HSync +Vsync
xrandr --addmode HDMI-A-0 "2880x1600_90.00"
xrandr --output HDMI-A-0 --mode 2880x1600_90.00
```

Tried with no success : 
```
xset dpms force off
sleep 1
xset dpms force on
```

On a special note, xrandr display `HDMI-A-0` while it's `/sys/class/drm/card0/card0-HDMI-A-1` that is the only HDMI output ? **There is also no edid for the O+**.
```
sudo echo "on" > /sys/class/drm/card0-HDMI-A-1/status
cat /sys/class/drm/card0/card0-HDMI-A-1/enabled 
disabled
```

johnr14, 2020-01-22:
DANG ! **got my headset to show** in LINUX !!! After all the trouble and all the help I got from nice people, thanks !

Here is why : 

```
In Fedora, the X configuration is determined automatically each time X is started. As a result, no xorg.conf file is created. In most cases, this works well and there is no need to manually specify X configuration.
```
What I think is that the **hmd takes longer to initialise** and when the xorg automatic configuration generator runs, the **hmd is not able to respond in time**.

This is a severe bug and **openhmd should prompt a error if it detects a headset but the screen is not configured**. That would be a lot more helpful.  

I created a Xorg.conf file and now can get information from xrandr and see my screen after : 
`xrandr --output HDMI-A-0 --prop --set non-desktop 0`

<details>
<summary>xrandr of odyssey+</summary>

```
HDMI-A-0 connected 1600x2880+5720+0 left (normal left inverted right x axis y axis) 2880mm x 1600mm
        EDID: 
                00ffffffffffff004ca34a141c00001f
                011b0103800000780a0dc9a057479827
                12484c00000001010101010101010101
                010101010101a0af40c4b04018607220
                a2004040b6000018147540c4b0401860
                7220a2004040b6000018c77340a2b040
                18606420a2004040b6000018000000fc
                005345433134344120202020200a01cb
                0203280065030c00100067d85dc40178
                8007755c12ca0107d709e2daca72cb4d
                9d4e0d3a90f38d04a0af40c4b0401860
                7220a2004040b6000018147540c4b040
                18607220a2004040b6000018c77340a2
                b04018606420a2004040b60000180000
                00000000000000000000000000000000
                00000000000000000000000000000058
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 1 0 0 0 0 0 0 0 1 0 0 0 0 0 0 
                0 1 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 74 
                supported: 74
        non-desktop: 0 
                range: (0, 1)
   2880x1600     90.00*+  60.00    60.00  
   1920x1200     90.00  
   1920x1080     90.00  
   1600x1200     90.00  
   1680x1050     90.00  
   1280x1024     90.00  
   1440x900      90.00  
   1280x800      90.00  
   1280x720      90.00  
   1024x768      90.00  
   800x600       90.00  
   640x480       90.00  
```
</details>

Mangmasta, 2020-02-28:
This is like a month late but I decided on a whim to check to see if I could get it to work, and without changing anything since I last tried a month ago, the screen on my Samsung Odyssey Plus is now showing up in my xrandr list without issue.  I am running Arch, maybe something updated within the past month that fixed this...