OpenHMD/OpenHMD #278: VR-Tek WVR2 drifts

TheOnlyJoey, 2020-02-02:
So, it seems like the Gyro offset variables might be wrong, or inconsistent for the WMR2.
When 0ing out all the values the one i tested barely drifts (though still constant to some axis).
I think we should re-evaluate this for the WVR2 and test with multiple devices to check if this is something we might need to enable in the firmware, or get from a config file.

For instance, the Oculus needs a feature set to enable sending corrected gyro and accel.

@l33tlinuxh4x0r @mdnelson8 Could you look into this?

mdnelson8, 2020-02-06:
I just retested this with my WMR2 and with the gyro offset values in the driver today I do get some small amount of drift over time, but for me it's far better than zeroing those values out. When I did the initial development I wasn't seeing any drift on my WMR2 with those values, but it was the height of summer. Now that it's the depths of winter (in the northern hemisphere at least), seeing some drift isn't too much of a surprise because when I was working out those values initially I noticed they were somewhat temperature dependent - cold start of the headset vs after a couple hours sitting atop my head.

Ideally, we'd read these offsets from the headset itself, but I didn't have much luck finding out which bytes they were, or even if they were present in the output we're seeing from the HMD at all. In the absence of this I like the idea of getting the offsets from a config file. That way each user can tweak them easily. I'm happy to work on this, but it might take a couple of weeks - free time is a little scarce at the moment.

TheOnlyJoey, 2020-02-06:
We should get them from the headset, or solve it in a different way, using a config for bias information is very hacky.

Probably will be good to go back to pcap dumping the initial communication (cold plug-in) and look for possible missed packages.

mdnelson8, 2020-02-07:
Yeah, I agree with that. We should definitely do things properly if we can. I'll skip the config file then :)

We're only using a handful of the bytes in the existing response packets today. I'm not sure what the bytes are that we're not using, but perhaps by examining dumps from your WVR2 as well as mine we can deduce something useful. It could also be that there are entirely different commands the HMD accepts that report useful info. The existing Windows driver is pretty bare bones and I'm pretty sure it simply relies on the HMD-provided quaternion, so there's probably more state space to explore for new commands too.

ChristophHaag, 2020-02-17:
I have the WVR2 in question now.

For reference:

this is the drift with the current values from the driver: https://streamable.com/7w39x

and when I set
```c
                    ofusion->gyro_offset[0] = 0;
                    ofusion->gyro_offset[1] = 0;
                    ofusion->gyro_offset[2] = 0;
```
then it looks like this: https://streamable.com/7pk5l