OpenHMD/OpenHMD #281: Interested in Extend Oculus positional implementation to other outside-in headsets

prabh27, 2020-02-21:
Hi, 

I am a Graduate Student at Carnegie Mellon University, pursuing a Masters in Electrical and Computer Engineering. My current research projects are in the field of Computer Vision.

Inventory Management Flow: In partnership with our industry partner in Thailand, we are working on real-time inventory analytics solutions to provide actionable intelligence to increase inventory turnover and improve product freshness. 

In this project, I am using Object detection and Object Tracking techniques to calculate the outflow and inflow of numerous products from the warehouses. Apart from using Deep Learning, we are also interested in providing a portable hardware solution in order to get scalability across all the warehouses in Thailand and to solve the challenge of slow or no internet connection at warehouses. 

Previously, I was working as a Software Developer for 3 years -- 2 years at Booking.com, Amsterdam and 1 year at Codenation Dubai/India. In the three years working in the industry, I worked on various projects ranging from Automatic Cloned Code refactoring to being the youngest core member of the NLP team at Booking.com. 

I think my experience will provide me a good start in understanding the details of this project, and coming up with a proper proposal. 

Can you please tell me more about this project?

TheOnlyJoey, 2020-02-21:
Ohai, 

So currently we have work being done on Positional tracking for the Oculus CV1 and DK2, the idea is to see if we can expand the work (introducing new filters, flexible camera setups, different pose estimation algorithms), and make it more general purpose.
Extending the tracking to PSVR or other outside-in tracked devices and setups would be the test case.

As mentioned in the idea list, it has to be dependency free (no opencv or big external libraries) and written in C99. OpenCV can be used during concept phase, but the algorithms need to be (re)implemented in OpenHMD and should be optimized for VR purposes. (low latency).

prabh27, 2020-02-22:
Sounds interesting! I'd like to ask a few questions to get a better understanding.

1) How much work has been done on Positional tracking for Oculus CV1 and DK2? 
2) Are you doing research on figuring out a new algorithm for positional tracking or using state-of-the-art algorithms? 
3) For the already existing algorithms, how much accuracy do they have?
4) The expansion of this work to add more features sounds like a low hanging fruit that can be achieved relatively quickly. We can try out GANs for new filters to try out a few artistic filters. 5) Different pose estimation algorithms can be integrated with the system as well, although the pose estimation algorithms have a lot of dependencies. As you mentioned, we can try to see the concept first, then try implementing it in OpenHMD. 
5) How would you prioritize the work mentioned? Adding more features or extending to other devices? I would say it can be a good way to understand the codebase by adding more features. Once I have clarity with the codebase, it will be easier to extend and reuse the existing code as well. 

I would break the project down into the following tasks:
1) Understanding the Positional tracking code for Oculus CV1 and DK2. 
2) Adding more features for Oculus CV1 and DK2.
     a) Introducing new filters - some basic filters. 
     b) Introducing artistic filters using GANs.
     c) Adding flexible camera setups (I am not sure about the steps involved in this).
     d) Trying different pose estimation algorithms. Comparing the accuracy and performance of each one of them, and also recording the popularity for each algorithm (user usage statistics).
3) Creating a concept for other tracking devices (Devices can be decided based on their usage or other metrics you consider important).
4) Implementing the positional tracking in OpenHMD.
5) Testing the implementation, writing documentation, test cases, a profiler for testing the speed.
6) Optimizing the code to have low latency and negligible errors.

How much time estimate would you give for each of the tasks? 
What are the other tasks you think I am missing in the plan mentioned above?

TheOnlyJoey, 2020-03-17:
As mentioned, it would be best if you send a email to info@thorworks.org to discuss the proposal further.

Currently we have DK2 and CV1 'done' with the exception of sensor fusion and some tweaking https://github.com/thaytan/OpenHMD/tree/rift-correspondence-search

It would be great if you would be improving the driver with new implementations, maybe the proposal from our idea list to work on a c99 library and implement that with the Oculus and/or new PSVR driver as final project. (https://github.com/OpenHMD/OpenHMD/wiki/GSOC-Idea's-(2020)#Implement_XR_optimized_filtering_for_sensor_fusion)

Please get in touch by email or drop by on IRC (irc.freenode.net #openhmd) to discuss this further!

poVoq, 2020-05-06:
So this never worked out?
See projects:
https://summerofcode.withgoogle.com/organizations/6210195455213568/

poVoq, 2020-05-11:
Has anyone looked into cooperating with OpenTrack:
https://github.com/opentrack/opentrack
They even have a SteamVR plugin: https://github.com/r57zone/OpenVR-OpenTrack