OpenHMD/OpenHMD #283: Quest & Link

se5a, 2020-03-06:
Are there any plans to attempt to support Oculus Quest via link?

I've just recently got a Quest, and being mostly Linux these days I'm feeling a little bit hampered, it looks like all the options to use the quest as a PC attached headset require the Oculus app on the PC, which is windows only. 

I might have a go to see if I can sideload some other android remote desktop options to try that angle, but I'd definitely like to try get the Link cable and have that working on Linux.

pH5, 2020-03-13:
Yes, do note though that Facebook are currently still changing the protocol without regard for backward compatiblity every other version.

se5a, 2020-03-15:
Excelent, if there's anything I can do to help let me know. 
My initial experiments trying to get the occulus pc software to run in proton by adding it to steam as a non steam game didn't get very far, didn't even get through the install stage, there's probably logs somewhere if i can find them.
experiments with remote desktop didn't quite work either, but I couldn't get my phone to work as a client anyway so there's other issues there.

thaytan, 2020-03-16:
I'd have been supremely surprised if that had worked - it needs the full Oculus system daemon and USB UVC stack to work in Wine

andy-d1969, 2020-03-28:
Maybe the [ALVR](https://github.com/polygraphene/ALVR) client could be used for the Quest side instead of the Facebook one.

se5a, 2020-03-28:
Oh nice find, I might have a look at that tomorrow.

andy-d1969, 2020-03-28:
There is also a [fork of ALVR](https://github.com/JackD83/ALVR) which is beeing more actively developed.

se5a, 2020-03-28:
Sooo looking into this a bit deeper, they're using some external drivers from FFmpeg which is cross platform so that shouldn't be too much of a problem, however it looks like they're using DX11 for the server side stuff, so kind of sunk there, with graphics stuff I'm waaay out of my depth.

ronthecookie, 2020-10-01:
I'm getting the Quest (Link) 2 soon so I will probably try and see if I can try and start getting it working with OpenHMD but it probably won't be here for another month or so.

andy-d1969, 2020-10-04:
ALVR is now being rewritten in Rust and will support Linux: https://github.com/JackD83/ALVR/projects/3#card-42919241
I've heard, they could use some help. Their Discord: https://discord.gg/KbKk3UM

ronthecookie, 2020-10-05:
Oh that's awesome! I'll see what I can do.

AmmarkoV, 2020-11-28:
By enabling developer mode one can develop a native application for the Quest platform from a purely Linux environment

You just need to add 
```
SUBSYSTEM=="usb", ATTR{idVendor}=="2833", ATTR{idProduct}=="0186", MODE="0666", GROUP="plugdev"
```
to /etc/udev/rules.d/51-android.rules to allow access to the quest for the adb utility

For example to stream from the Quest2 to the PC you can use 
```
adb exec-out "while true; do screenrecord --bit-rate=2m --output-format=h264 --time-limit 180 -; done" | vlc --demux h264 --h264-fps=60 --clock-jitter=0 -
```
I suppose making a .apk "driver" application running on the quest is the most  feasible approach to make a front-end that given a fast USB/Wifi connection could plug-in to the rest of the OpenHMD stuff

Example native application that can be compiled through this docker image  :
https://github.com/BrainSlugs83/DockerOvrSdk
https://github.com/DrBeef/QuakeQuest

rhysperry111, 2021-03-14:
Sorry to bump this thread, but I was looking into getting a quest and OpenHMD support would be extremely useful. What is the current status of this?

leder11011, 2021-03-15:
I was told to watch the social nets for openhmd supporting quest 1: I did this for half a year - nothing new! ALVR seems to be work in progress for Linux, cp this link:

[ALVR](https://github.com/alvr-org/ALVR/wiki/Linux-Support-development-progress)

ronthecookie, 2021-03-16:
@leder11011 The current ALVR Linux support is a proof-of-concept by @xytovl however they're working on a better approach now that should be more portable.

TheOnlyJoey, 2021-03-16:
Once i get my usb bulk patches done for our HIDAPI fork, we can start merging the positional tracking from the Quest with Link cable. Video still needs figuring out for a fully working driver though, the codec seems quite difficult to figure out.

cpyarger, 2021-03-18:
Need someone to do some testing?

alexfreyre, 2021-06-30:
Hello everybody! I am very excited about using Quest 2 on Linux! 

I have a question, are the advances so far by cable or wifi?

wizardwatch, 2021-06-30:
> Hello everybody! I am very excited about using Quest 2 on Linux!
> 
> I have a question, are the advances so far by cable or wifi?

I would suggest looking into ALVR. https://github.com/alvr-org/ALVR

TheOnlyJoey, 2021-06-30:
The progress so far is through link cable, since that is the easiest
approach.
This is because all drivers so far in OpenHMD have been implemented through
HID/USB data.
Currently patches are made to our hidapi fork that would make it possible
to extract the positional data.
From there it is just a case of reverse engineering the video protocol.

Wireless streaming and such is not really part of OpenHMD yet, but is being
looked at.

alexfreyre, 2021-06-30:
Thanks for the answer. Good to know that there is some progress through link cable.

 I'm a FOSS advocate and I'm currently developing VR apps with Godot on Linux via OpenHMD, so far I''ve used PSVR and it works well, but I'm looking for a new headset and I'm considering to buy the Quest 2 because its great resolution, features and price.

But I've some questions in order to make the right choice.
Is the **Oculus Quest 2** already supported? 
What is the current state?
Does it works now via cable?
Is supported the hand tracking feature?
What about positional tracking?

AndKe, 2022-06-18:
I am searching for the same as @alexfreyre   ... Is Oculus Quest 2 somehow supported?  - I would love to use it for X-Plane.

janat08, 2022-07-10:
I was hoping to use wifi instead.

shinyquagsire23, 2022-11-07:
I've made pretty rapid progress in my repo at https://github.com/shinyquagsire23/xrsp_tests, so far I have pose data and camera images streaming from a Quest Pro (and I think mic data? I haven't attempted to decode it). And as far as I can tell, streaming video is just a matter of H265 encoding. Though I'm not 100% on how it works, I have a packet dump of the Slice 0..15 pkts but no idea how they segment the display. At least I assume that's what 'slice' is referring to.

thaytan, 2022-11-08:
At some point I looked at Quest Link capture, and in that one it was sending the frame as 5 parallel video streams, each 2016 pix wide by 224 pixels high. At the time, that was H.264, because they hadn't added H.265 yet - and of course the protocol may have changed since then.