OpenHMD/OpenHMD #288: Segmentation fault in rift.c/get_device_list with VR-Tek WVR2 on Ubuntu

fluffrabbit, 2020-04-19:
After compiling OpenHMD from master with cmake in debug mode, I ran the simple example with the headset plugged in and got a segmentation fault. Here's my backtrace from `ddd`:

```
in main () at simple.c:58
in ohmd_ctx_probe () at openhmd.c:120
in get_device_list () at rift.c:1081
wcscmp () at wcscmp.S:207
```

The relevant line in `rift.c` is:

`if((wcscmp(cur_dev->manufacturer_string, L"Oculus VR, Inc.") == 0) &&`

I figure something is wrong with `cur_dev->manufacturer_string`. The relevant line from `lsusb` is:

`Bus 001 Device 007: ID 2833:0001  `

It looks like the name is empty. I hope that isn't the source of the segfault.

I should also note that I have installed OpenHMD from apt and the experimental SteamVR drivers from Valve. I also added their udev rules from [here](https://github.com/ValveSoftware/steam-devices/blob/master/60-steam-vr.rules) but didn't mess with OpenHMD udev rules. `xrandr --props` reports the headset as `non-desktop`.

thaytan, 2020-04-19:
I was just hitting this last night suddenly too (on Fedora). I don't know why.

hid_enumerate() comes from libhidapi - of which there are 2 variants on Linux libhidapi-libusb and libhidapi-hidraw with 2 backends. It's supposed to fill an array of all devices that match the Rift USB device VID:PIDs

The default should be using libhidapi-libusb, so the question is why libusb didn't fill out the manufacturer string field

fluffrabbit, 2020-04-19:
>I was just hitting this last night suddenly too (on Fedora). I don't know why.
Spooky!

>there are 2 variants on Linux libhidapi-libusb and libhidapi-hidraw
After reading about it, I assume libhidapi-libusb is the superior option for USB devices.

>why libusb didn't fill out the manufacturer string field
Either because I set up my system wrong or it's a $30 headset from China.

So, I have two problems:

1. I don't know if I've set up my headset correctly or not. (Still haven't found a SteamVR game that will run on my system, and OpenHMD is obviously where I am now.)

2. In case it's the manufacturer's fault, or just bad voodoo magic, I think that instead of segfaulting, OpenHMD should detect a null string output a helpful diagnostic message, continuing as usual and seeing what happens.

thaytan, 2020-04-20:
I can't reproduce the crash today - the manufacturer_strings are set every time I try.

Maybe this only happens when the device is freshly plugged in and still being probed by the kernel?

If so, we should just add NULL checks in both places.

fluffrabbit, 2020-04-20:
I connected the headset, left and came back 20 minutes later, ran the simple example, and it segfaulted as before. `lsusb` still reports no manufacturer string.

Even though our experiences are different, I would recommend null checks just to be on the safe side.

thaytan, 2020-04-20:
lsusb reporting no manufacturer string is the interesting bit - that really does point at the USB stack or device being the problem.

I see 5 places in OpenHMD currently accessing the manufacturer string. Do you want to make an MR?

```
src/drv_deepoon/deepoon.c:              if (wcscmp(cur_dev->manufacturer_string, L"DeePoon VR, Inc.")==0 &&
src/drv_htc_vive/vive.c:        dump_info_string(hid_get_manufacturer_string,
src/drv_nolo/nolo.c:    if (wcscmp(device->manufacturer_string, L"LYRobotix") != 0) {
src/drv_oculus_rift/rift.c:                     if((wcscmp(cur_dev->manufacturer_string, L"Oculus VR, Inc.") == 0) &&
src/drv_vrtek/vrtek.c:        if (wcscmp(cur_dev->manufacturer_string, L"STMicroelectronics")==0 &&
```

fluffrabbit, 2020-04-20:
>that really does point at the USB stack or device being the problem.
I know, right? Or how I've set it up.

>I see 5 places in OpenHMD currently accessing the manufacturer string. Do you want to make an MR?
Yes, but not quite yet. I think it would be a good idea to output diagnostic information, but I'm not sure what OpenHMD's policy is for printing to stderr, and I'm still a little foggy on what the end user would deduce from the computer telling them there's no manufacturer string.

thaytan, 2020-04-20:
here's a thought: the manufacturer_string is a translated string. I wonder if a non-C/en locale leads to the string being NULL?

Can you attach the lsusb -v output for your device?

fluffrabbit, 2020-04-20:
Interesting idea. I piped it into this file via `lsusb -d 2833:0001 -v > lsusb-wvr2.txt` at which point lsusb said `Couldn't open device, some information will be missing`.
[lsusb-wvr2.txt](https://github.com/OpenHMD/OpenHMD/files/4504440/lsusb-wvr2.txt)

thaytan, 2020-04-20:
You get that message unless lsusb is running as root, but I don't think it matters here. iManufacturer=1 in the lsusb output, so there *should* be a string descriptor available for retrieval.

I think the device is just troublesome.

The point of these string checks was that the WVR devices decided they'd use Oculus' USB ids, but don't work like a Rift, so we needed a way to distinguish them in the drivers. If we can't rely on the manufacturer string being available though, we might need to come up with another way of distinguishing them

fluffrabbit, 2020-04-20:
My thought on this discrimination are pretty simple: The cheaper the headset, the less likely it is to output useful information. WVR1/2/3 are the cheapest on desktop, so I would default to a generic VR-Tek driver if it has an Oculus ID but no string.

fluffrabbit, 2020-04-20:
> That sounds OK - except that somehow I hit this segfault yesterday
> with a Rift S :-/

Dunno what to tell you.

thaytan, 2020-04-20:
I'll see if I can reproduce it again somehow.

fluffrabbit, 2020-05-04:
@thaytan I take it you couldn't reproduce the issue with your Rift S. How would you suggest I proceed?

thaytan, 2020-05-04:
Correct, I haven't seen it again since. 

@pH5 pointed out commit cdf05814456ed8cab113bf9460d3ad9d1c1cf8db should already have added a NULL check to rift.c though.

fluffrabbit, 2020-05-04:
I pulled from master weeks after that commit when I encountered the bug. To be worth anything, the null check needs to exist everywhere that `wcscmp` is used, or strings need to be checked preemptively before multiple calls to `wcscmp`. That's what I meant by "It's pointless to keep kicking this can down the road".

thaytan, 2020-05-04:
agreed - I think we should wrap wcscmp and do a NULL check in it

fluffrabbit, 2020-05-04:
I like that idea. Here's a drop-in replacement for `wcscmp`:

```c
int WideStringCompare( const wchar_t *lhs, const wchar_t *rhs ){
	if( !lhs || !rhs )
		return -1; // A stand-in for a general mismatch.
	return wcscmp( lhs, rhs );
}
```

Here's a more idomatic version that returns `true` if match or `false` if mismatch:

```c++
bool WideStringMatch( const wchar_t *a, const wchar_t *b ){
	if( !a || !b )
		return false;
	return wcscmp( a, b ) == 0;
}
```

You're not doing binary trees or regex or anything so I'm inclined to go with the latter approach.

fluffrabbit, 2020-05-05:
@thaytan I'm preparing a PR. No more segfault, but when I run the simple example I just repeatedly get this:

```
rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 
```

I'll PR it but I'm not sure how to proceed after that.

thaytan, 2020-05-06:
Thanks! 

Output like that usually means the device wasn't detected, and you are getting data from the fallback null device. Is your WVR2 in the list of devices printed when the simple example starts? If so, it's something wrong with the driver itself - but I don't have a WVR2 to test with - so someone else will need to help there (and we should move to a new issue report)

fluffrabbit, 2020-05-06:
>Is your WVR2 in the list of devices printed when the simple example starts?

Unfortunately, yes (in a way). I'm attaching the output here.

>we should move to a new issue report

Not a bad idea; maybe you can suggest how to word it.

[openhmd-simple-wvr2-out.txt](https://github.com/OpenHMD/OpenHMD/files/4585545/openhmd-simple-wvr2-out.txt)

thaytan, 2020-05-06:
That output isn't showing your WVR2. The "external device" driver is a different thing - I think it's for an external process to send fake motion info for remote control of a VR app.

I'm not sure what the output should look like for a WVR headset, but it should mention the WVR string somewhere. For the Rift CV1, it looks like:

```
num devices: 8

device 0
  vendor:  Oculus VR, Inc.
  product: Rift (CV1)
  path:    0001:0014:00
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  Oculus VR, Inc.
  product: Rift (CV1): Remote Control
  path:    0001:0014:00
  class:   Controller
  flags:   00
    null device:         no
    rotational tracking: no
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  Oculus VR, Inc.
  product: Rift (CV1): Right Controller
  path:    0001:0014:00
  class:   Controller
  flags:   14
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    yes

device 3
  vendor:  Oculus VR, Inc.
  product: Rift (CV1): Left Controller
  path:    0001:0014:00
  class:   Controller
  flags:   0c
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no
....
```

fluffrabbit, 2020-05-06:
I see. OpenHMD didn't detect my device. The Oculus Rift device ID shows up without a name in lsusb. Not sure how to proceed. I'll try booting into a non-SteamVR kernel and see what happens. Probably the same result. Hardware devices are always a little finicky on Linux. I'm recalling my early struggles with Android and the udev solution I employed.

thaytan, 2020-05-06:
The most common problem is permission on the USB devices - there's a wiki page about that https://github.com/OpenHMD/OpenHMD/wiki/Udev-rules-list

fluffrabbit, 2020-05-06:
I don't know how to read quaternions, but they are definitely moving now! I knew it must have been something stupid like that. Thanks! Now I'll try to get something to show up on the goggles.

fluffrabbit, 2020-05-06:
I tried the OpenGL example, but it's just a window and the eye holes are dark. I believe the WVR2 is a "real mode" headset, so dragging the window over may not be a viable strategy. Is there a newer example somewhere?

thaytan, 2020-05-06:
For my OpenHMD testing, I usually just use the opengl example - which as you say, pops up a window rather than displaying on the HMD. My goal is usually just to test tracking and positioning though, so that doesn't matter.

For a real use, you need to look to an app or lib that can use OpenHMD, like the Monado OpenXR runtime, the Godot plugin, or SteamVR.

I do think a standalone demo that looks up and configures output to the correct display device would be useful, but I don't know of one that works.

fluffrabbit, 2020-05-06:
As a game developer who is weeks away from the Steam launch of a non-VR game, I'm looking to the future for my next project, and the lack of a good example makes me a little nervous about using OpenHMD directly.

Monado looks interesting, but I'm not seeing it in Apt, the documentation looks fairly casual, and I don't know what the OpenXR API brings that OpenHMD does not. The 1.0 spec just came out and the writings seem fixated on 0.9 and the instability of the API. Not sure how to proceed with that.

I used to contribute to Godot, and I suppose I can set it up just to test the headset, but I won't be using Godot in my projects.

The [url=https://github.com/ChristophHaag/SteamVR-OpenHMD]SteamVR OpenHMD thing[/url] looks the worst of all. So many steps, so many bugs.

What looks the most appealing to you?

PR #291 (2020-08-14): Fix segfaults with null strings passed to wcscmp
Fixes #288
closed 2020-08-14 by TheOnlyJoey