OpenHMD/OpenHMD #292: No display on WMR headset/steamVR not detecting (on windows 10)

herosilas12, 2020-05-20:
I downloaded OpenHMD-0.3.0-msvc2017.zip
and ran the .exe in that folder and it did detect the headset but there was no display and then I tried to start up steamVR but it did nothing
I want to contribute to the project but I dont have any experence coding so I can send diagnostic/tracking info if you can assist me
thank you :)

herosilas12, 2020-05-20:
now detects the headset but I get this error
[image]

herosilas12, 2020-05-20:
uninstalled windows mixed reality portal and now it detects it without the error. however I still get "failed to stop OVRService, try running as admin" even though I don't have the oculus software installed or running

TheOnlyJoey, 2020-05-20:
I think you might be confused, OpenHMD itself does not do anything when not used in a application. It seems you are trying to get the WMR running with OpenHMD for SteamVR?
https://github.com/ChristophHaag/SteamVR-OpenHMD Is the project you need then, keep in mind though that it currently requires basic software compilation knowledge, and Windows support is undergoing some patches.

closed 2020-05-20 by TheOnlyJoey