OpenHMD/OpenHMD #295: HP Reverb (wmr) display fails to be detected, usb connection works

nismad01, 2020-05-28:


letoram, 2020-05-28:
As far as we know the reverb needs some extra wakeup command sent for the display to become 
active.If you have a second machine with windows installed you can verify that by having the usb hooked up to the windows machine, and the display to the linux system.

nismad01, 2020-05-28:
Wow. I will try that ASAP. I wonder if any crypto is involve. Otherwise replaying the commands might work, in an ugly way.

nismad01, 2020-05-29:
I can confirm that connecting the usb cable to a Window machine works. The Reverb headset powers up and appears as yet another monitor connected to the Xubuntu laptop's external GPU.

The Xubuntu's desktop can be seen on the HMD, -but obviously not split up for stereo viewing. Fish-eye-like distortion is quite apparent though, so correction for that will clearly be needed.

I will try use wireshark to capture the initial exchange of USB data between Windows and the Reverb, and see if I can make any sense of it, and/or make the Xubuntu box replay it. I'm hoping it does not involves challenge-response or other methods of locking it to Windows, since I'm no crypto expert.

I wouldn't be surprised if the Reverb could be physically powered up by hw-hacking it, but I prefer not to void the warranty just yet.

-On a side note, I realize that getting full inside-out positional tracking based on CV processing of the surrounding walls, floor and furniture is probably not possible within a reasonable time-frame. But since I only intend to use the HMD when seated, I have considered a much more limited approach based on just identifying the four corners of the computer screen, and then use their relative positions to calculate translational movement of the HMD. Obviously, it would one work when the screen is within view of the headset cameras, but just having a little freedom to control the parallax would be much better than none at all. And I would expect this to be an achievable realtime CV project.

letoram, 2020-05-30:
I use it with [safespaces](https://github.com/letoram/safespaces), but so far only through the windows bootstrap method, it's in the pile of hw- reversing projects, but quite far down. There is a cap @ [reverb.pcap.xz, 30mb, forumfiles.com](https://forumfiles.com/z9ubK43fo6/reverb.pcap_xz) - as you see there are multiple devices involved, a suspect was/is 03f0:0c6a (3.11.0) though no luck poking it.

kflansburg, 2020-05-30:
[reverb.pcap.zip](https://github.com/OpenHMD/OpenHMD/files/4706216/reverb.pcap.zip)

I'm also very interested in getting this to work. I have attached a pcap of the `03f0:0c68` device when I plug in my Reverb and then launch the WMR portal.

nismad01, 2020-06-03:
Sorry for not responding earlier. I have been to busy with some of the coding work that actually feeds me. Thank you, letoram and kflansburg, for helping out. I am really on to it, and will take a deep dive into it as I can. I am currently chained to my work laptop, so I havn't downloaded your files yet.
Just to get up to speed with your latest suggestions: Are these "suspect" (USB) devices you are talking about the potential power controller for the Reverb display? -Being new to the OpenHMD project, I may lack a bit of the overall context and terminology, so I may need the wording a bit more explicit here to begin with.

kflansburg, 2020-06-03:
My Reverb presents several USB devices, and the two most likely ones are Microsoft HoloLens Sensors, and HP Reverb. Both appear to be HID devices, and I would guess since this functionality doesn't appear in other WMR devices that the HP Reverb device is where the power controller is. As you can see from the PCAP, there is very little information sent to the HP Reverb device right at startup, and I don't believe I see any messages transferred when I closed the WMR portal, so it is possible that this isn't the device that controls power. 

I did notice that if I start the Reverb in Windows and then reboot right into Linux, that the Reverb actually stays active and this is the only time I have seen it appear in `xrandr`.  I would like to help reverse this as well, but have very little experience with USB / HID devices so it will take me some time to get familiar with it.

kflansburg, 2020-06-19:
I have not had much time to work on this, but have found some additional information. In particular there is another device (with Reverb in the name) which does not appear to be a HID device which may be worth inspecting the pcaps of.

| HID Vendor | HID Product | Vendor ID | Product ID | `lsusb` Name |
|------------------|------------------|---------------|----------------|---------------------|
| Microsoft     | HoloLens Sensors | 1118 / 0x045e | 1625 / 0x0659 | Microsoft Corp. | 
| Quanta Inc. | QHMD A85V | 1008 / 0x03f0 | 3178 0x0c6a | HP, Inc |
| | | 0x04b4 |  0x6504 | Cypress Semiconductor Corp. HP Reverb VR Headset |

I'm including the detailed `lsusb` of  the last device. The remote wake-up attribute seems very interesting. 

```
Bus 004 Device 002: ID 04b4:6504 Cypress Semiconductor Corp. HP Reverb VR Headset
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x04b4 Cypress Semiconductor Corp.
  idProduct          0x6504
  bcdDevice           50.00
  iManufacturer           1 HP Inc.
  iProduct                2 HP Reverb VR Headset
  iSerial                 3 REDACTED
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             1
  wHubCharacteristic 0x0004
    Ganged power switching
    Compound device
    Ganged overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.3 micro seconds
  wHubDelay          1507 nano seconds
  DeviceRemovable    0x02
 Hub Port Status:
   Port 1: 0000.0203 lowspeed enable connect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             REDACTED
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

kflansburg, 2020-06-19:
I believe the Cypress device is a Hub device for the various Microsoft devices. I have attempted to power on various USB devices using the procedures discussed [here](https://www.kernel.org/doc/html/v4.18/driver-api/usb/power-management.html), but have had no success.

kflansburg, 2020-06-20:
I went ahead and patched the WMR driver to ignore the unknown messages, and can properly load the device config JSON. Still no luck turning on the display though. 

https://github.com/kflansburg/OpenHMD/blob/c64cf2f59aba22b8220c20c0b13ee2a09c380dee/src/drv_wmr/wmr.c#L205

```
opening device: 0
[II] 045e:0659 000a:0003:02

[II] opening

[EE] Unrecognized Command
[EE] Unrecognized Command: 23
[EE] Unrecognized Command: 0
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 94
[EE] Unrecognized Command: 4
[EE] Unrecognized Command: 91
[EE] Unrecognized Command: 6
[EE] Unrecognized Command
[EE] Unrecognized Command: 23
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 94
[EE] Unrecognized Command: 4
[EE] Unrecognized Command: 91
[EE] Unrecognized Command: 6
[II] Read 10725-byte config data

[II] Model name: HP Reverb VR Headset VR1000-2xxx

[EE] Found display height 2160

[EE] Found display width 4320

resolution:              4320 x 2160
hsize:                   0.103812
vsize:                   0.051905
lens separation:         0.063000
lens vcenter:            0.025953
left eye fov:            1.658063
right eye fov:           1.658063
left eye aspect:         1.000000
right eye aspect:        1.000000
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
control count:           0
controls:
```

kflansburg, 2020-06-20:
@nismad01 Based on the PCAP I posted above, I was finally able to determine the initialization sequence. I have created a pull request with the new functionality. 

There is still an issue where I get an error reading the device config, however it must be some sort of race condition because if you try a couple of times, it will eventually work. 

Error:
```
opening device: 0
[II] 045e:0659 0004:0003:02

[II] opening

[EE] Failed to issue command 08: 17 00 01

[EE] Could not read config from the firmware
```

Successful:

```
opening device: 0
[II] 045e:0659 0004:0003:02

[II] opening

[II] Read 10725-byte config data

[II] Model name: HP Reverb VR Headset VR1000-2xxx

[EE] Found display height 2160

[EE] Found display width 4320

[II] Detected HP Reverb
[II] Initializing Reverb.
```

yackaro, 2020-06-30:
I seem to be having a similar issue with the Samsung Oddysey+. It seems the work has been done, but the work is only being applied if the product ID of the Reverb is detected. I believe that this same fix likely also applies in my case. My Oddysey+ is on the latest firmware available, 1.0.2, as reported by the Samsung VR setup app in Windows 10. Could it have something to do with newer firmware revisions implementing this check, or am I just being less than smart about setting up OpenHMD on my copy of KUbuntu 20.04? I am willing to allow remote desktop access to my machine, if it will help. I'm not experienced in how to modify code to interface with various devices, so I don't expect to be much help otherwise.

kflansburg, 2020-07-01:
@yackaro There may be something similar that needs to be done for the Odyssey, butI doubt this patch will work, it is hard coded to write to the HP USB device. You should probably open a new issue, and I can try to help you out there. For now, can you run `lsusb -vt` with the Odyssey plugged in on kubuntu and see if there are Samsung USB devices?

yackaro, 2020-07-01:
I do indeed have the Samsung USB devices present, as that is one of the things I checked when troubleshooting it initially.
The relevant section of lsusb:

> /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        ID 05e3:0610 Genesys Logic, Inc. 4-port hub
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            ID 2109:2210 VIA Labs, Inc. 
            |__ Port 2: Dev 5, If 0, Class=Audio, Driver=snd-usb-audio, 480M
                ID 04e8:7084 Samsung Electronics Co., Ltd 
            |__ Port 2: Dev 5, If 1, Class=Audio, Driver=snd-usb-audio, 480M
                ID 04e8:7084 Samsung Electronics Co., Ltd 
            |__ Port 2: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 480M
                ID 04e8:7084 Samsung Electronics Co., Ltd 
            |__ Port 2: Dev 5, If 3, Class=Human Interface Device, Driver=usbhid, 480M
                ID 04e8:7084 Samsung Electronics Co., Ltd 
            |__ Port 3: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                ID 04e8:7312 Samsung Electronics Co., Ltd 
        |__ Port 2: Dev 4, If 0, Class=, Driver=, 12M
            ID 1d5c:7102 Fresco Logic 

I will be looking into running wireshark on the USB hub in windows, though, and will report any results I get back in a new issue thread. I posted here because it seems to be a similar root cause and will probably require a similar, or even identical, fix.

kflansburg, 2020-07-01:
Sounds good. If you can provide a PCAP of that device while Windows powers it up, I can take a look and see if a similar patch can be used.

spiderbit, 2021-09-24:
Well did test it rebooted after deinstalling the openhmd version from you and build / installed the version from @thaytan rebooted and still nothing in xrandr.

Any idea which log file I should find some related error? nothing in xorg.conf that catches my eye or dmesg... 
Is that normal:
```
config 1 has an invalid interface number: 3 but max is 2
config 1 has no interface number 0
```

[system.log](https://github.com/OpenHMD/OpenHMD/files/7222020/system.log)

kflansburg, 2021-09-24:
> Well did test it rebooted after deinstalling the openhmd version from you and build / installed the version from @thaytan rebooted and still nothing in xrandr.
> 
> Any idea which log file I should find some related error? nothing in xorg.conf that catches my eye or dmesg...
> Is that normal:
> 
> ```
> config 1 has an invalid interface number: 3 but max is 2
> config 1 has no interface number 0
> ```
> 
> [system.log](https://github.com/OpenHMD/OpenHMD/files/7222020/system.log)
> [PKGBUILD.txt](https://github.com/OpenHMD/OpenHMD/files/7222128/PKGBUILD.txt)
> 
> Did makepkg and makepkg --install (the file must be without txt github don't allow uploads of files without endings)

It sounds like we may have different versions of the headset, but the problem for me was always that the headset display was not powered on. So I never saw any errors, I would just visually inspect the display and see if the backlight was on (good) or not (bad). 

This script should get the display to turn on, and I may have had unplug/replug the HDMI cable or reboot the system _after_ the display was on to get it recognized. I always manually ran the script, not as a system service. 

Do you have the stdout/stderr from the script itself? It is possible that it is not detecting the HP device since it is a different version (and therefore might have a different identifier). 
If you are using my script (im not sure if thaytan kept the same logs), you should see:
```
Detected HP Reverb
```

Regarding the loglines you shared, I'm not sure if those are normal, that would probably be a question better answered by the maintainers.

spiderbit, 2021-09-24:
What script are you talking about? I just install the driver, is there a bash script included? If so what is the name? Btw at least the Reverb v1 has a Displayport not a Hdmi port...

kflansburg, 2021-09-24:
> What script are you talking about? I just install the driver, is there a bash script included? If so what is the name? Btw at least the Reverb v1 has a Displayport not a Hdmi port...

You are right, Displayport. 

I was simply compiling `examples/simple.c`. I never tried anything more complex than that.

spiderbit, 2021-09-24:
Well the simple just throws out some numbers that change when I move around the device so it recognises the movement, but that does not make it visible with xrandr.

spiderbit, 2021-09-24:
I only see "kernel: usb 3-3: Product: Hp Reverb VR Headset" are we talking about journalctl or dmesg?

kflansburg, 2021-09-24:
> I only see "kernel: usb 3-3: Product: Hp Reverb VR Headset" are we talking about journalctl or dmesg?

I was never looking at journalctl or dmesg. I think you need to verify that OpenHMD is detecting that the headset is HP and applying the startup sequence defined in the patch. If it isn't, then the display is not on, and it will not show up in xrandr.

spiderbit, 2021-09-24:
And in what logfile or through what command do I trigger that, I have installed the modified version from @thaytan that should have the patch included right? So Openhmd just starts in the background right, there is no openhmd.sh so what is it a X driver so in xorg.0.log or where would I expect this message to see?

spiderbit, 2021-09-24:
```
device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
```
Is that the problem with openhmd-simple-example?

kflansburg, 2021-09-24:
> And in what logfile or through what command do I trigger that, I have installed the modified version from @thaytan that should have the patch included right? So Openhmd just starts in the background [...]

The patch code was invoked when I ran the `simple` example, and the logs should go to stdout/stderr. If you are not seeing:

```
[II] Detected HP Reverb
[II] Initializing Reverb.
```

Then I do not believe it is running. I don't recall if I had to do something to enable logging.

spiderbit, 2021-09-24:
Sometimes I get that instead:
```
[EE] Failed to issue command 08: 17 00 01
[EE] Could not read config from the firmware
```
and I always see those a bit later:
```
[EE] unknown message type: 23
[EE] unknown message type: 23
```
I got some detecting "4300 x 2160" or so just before instead of this Failed to issue commend and firmware but that made it still not visible to xrandr. Don't know what changed now that it went to this failed to issue thing, even rebooting hard my pc don't brings that back.

spiderbit, 2021-09-24:
Ahh I see how to reproduce the other message have to start the simple tool as root:

```
opening device: 0
[II] 045e:0659 0004:0003:02
[II] opening
[II] Read 10751-byte config data
[II] Model name: HP Reverb VR Headset VR1000-2xxx
[EE] Found display height 2160
[EE] Found display width 4320
resolution:              4320 x 2160
hsize:                   0.103812 
vsize:                   0.051905 
lens separation:         0.063000 
```

spiderbit, 2021-09-24:
Well the original poster made the comment that he had the same error message:

```
[EE] Failed to issue command 0b: 17 00 00
[EE] Could not read config from the firmware
```
funny enough I get past that, but the only way I get past that it seems consistently, is if it breaks for whatever reason, is to:
install your branch @kflansburg
deinstall it
install @thaytan version
run simple example as root

then I get not this error but that it detects the resolution like posted in my last comment, that it seems that I have to be root to get that far, suggest some user/rights issues?

thaytan, 2021-09-25:
> then I get not this error but that it detects the resolution like posted in my last comment, that it seems that I have to be root to get that far, suggest some user/rights issues?

Yes, needing root privileges implies that you are missing udev rules to give your user account access to the right USB devices. Take a look at https://github.com/OpenHMD/OpenHMD/wiki/Udev-rules-list and in particular the section about the xr-hardware package

spiderbit, 2021-09-25:
I have the xr-hardware package installed, so I wonder what's the problem.

spiderbit, 2021-09-25:
So I run again some tests, I removed the xr-hardware and added the ids manually, but I think that was not the problem, my previous observation that it only works as root was wrong, it just randomly got success as root and not as root because I made not many tests.

So from 10 tests as user I get 3-4 times the notice with resolution and the rest of the times the failed comment, that is true with both versions. Now I am unsure what to look next xrandr still shows nothing useful.

spiderbit, 2021-09-25:
But even if it succeeds that part:

```
[II] Detected HP Reverb
[II] Initializing Reverb.
```
Never comes

spiderbit, 2021-09-25:
hmm I see 1 difference @kflansburg had:
`[II] Read 10725-byte config data`
while I have:
`[II] Read 10751-byte config data`

Does that imply that I have a newer firmware?

spiderbit, 2021-09-25:
Ok found the problem I did not use the correct branches in the PKGFILES
here again the 2 that work if somebody wants to use them:
[PKGBUILD-thaytan.txt](https://github.com/OpenHMD/OpenHMD/files/7230200/PKGBUILD-thaytan.txt)
[PKGBUILD-kflansburg.txt](https://github.com/OpenHMD/OpenHMD/files/7230201/PKGBUILD-kflansburg.txt)
That makes Extended mode possible, sadly I struggle with direct mode with steamvr now...

blurymind, 2022-04-05:
interested to see if this gets support :) any progress since last year?

spiderbit, 2022-04-05:
@blurymind I tried a session on the monado irc channel where you could go into, too.
The problem seems with the steamvr emulation part, the openhmd part seems to work, but with this driver the steamvr driver no matter if monado or this directly is unusable.

I assume the people that wrote the openhmd driver never tested steamvr beacuse it's not working btw the same is true for G1 and G2.