OpenHMD/OpenHMD #296: Support Hypereal Pano ?

Xanadu-Zen, 2020-05-29:
Hypereal Pano is a multi camera IR blob tracking based VR headset, this headset start selling from June 2017, Include dual-camera or triple-camera, sens controller and the HMD. dual camera version provide 240° and triple camera provide 360° tracking.
The manufacturer also released compatible tools on Windows to work with SteamVR at that time. but after January 2018, the manufacturer was transformation to other industry and no longer to support this headset.
With tons of updates from SteamVR, the compatible tools are not work with the latest SteamVR(can not detect the key press event in game ).
Now this VR headset can only work perfectly with old version SteamVR (such as macos_beta, 1.3.23).
Does anyone interested to support this devices? For necessary, I can sponsor a triple-Camera headset with the latest client drivers from manufacturer and old version SteamVR macos_beta for reverse engineering.

Xanadu-Zen, 2020-05-29:
The device looks like this
[image]
and it's poweron
[image]

thaytan, 2020-05-29:
Neat! I've never heard of it. I wonder how much of the CV1 work will translate

Xanadu-Zen, 2020-05-29:
This manufacturer open source they frist genration product HR-MK1-HMD (pano in photos is HR-MK2-HMD), but  the MK1 is a laser positioning based devices, that might not help for MK2

Xanadu-Zen, 2020-05-31:
My package is ready to go, please give me shipping address !

thaytan, 2020-06-01:
I'm already going to be busy with Rift CV1 and Rift S quite a while - so if anyone else wants to work on this headset they should say so

TheOnlyJoey, 2020-06-01:
@Xanadu-Zen please email me at info@thorworks.org will make sure we get a basic driver going, and hope that we can reuse part of the DK2/CV1 implementation!

Xanadu-Zen, 2020-06-08:
@TheOnlyJoey 
I have sent the email yesterday，hoping for your reply!

naratteu, 2020-09-03:
I also need an open source driver for hypereal pano.
I hope the link below will helpful.
https://github.com/hypereal-opensource

shockla, 2020-11-24:
I just bought a used Hypereal Pano, Express on the way (a full set, seems to be same as picture above)
I heard that some chat group have offline driver installer (size: ~750MiB)
I'm trying to get it.

closed 2020-11-24 by Xanadu-Zen

kaibergman, 2021-11-18:
I have just received an MK2 headset in the mail without any lighthouses or controllers. Would anyone be able to send me the drivers?

Xanadu-Zen, 2021-11-18:
> I have just received an MK2 headset in the mail without any lighthouses or controllers. Would anyone be able to send me the drivers?

Xanadu-Zen, 2021-11-18:
> I have just received an MK2 headset in the mail without any lighthouses or controllers. Would anyone be able to send me the drivers?

I have upload the driver to Google drive, here is the link

kaibergman, 2021-11-18:
Thank you, i'm downloading it now. Do you know of any way of getting some motion functionality without the official lighthouses ?

kaibergman, 2021-11-19:
The Driver installed fine however now I can't get past the set up screen as it isn't detecting the HDMI. I believe it's because my hdmi output is connected to my igpu and not the dedicated graphics card. Is there no way past this? I suppose the hypereal software doesn't support my kind of hardware.

Xanadu-Zen, 2021-11-19:
> Thank you, i'm downloading it now. Do you know of any way of getting some motion functionality without the official lighthouses ?

no，this hypereal driver can only support IR camera in headset kit 

> The Driver installed fine however now I can't get past the set up screen as it isn't detecting the HDMI. I believe it's because my hdmi output is connected to my igpu and not the dedicated graphics card. Is there no way past this? I suppose the hypereal software doesn't support my kind of hardware.

Driver need  find IR camera and controller connected to complete environment setup. The HDMI on headset should plug into dedicated AMD/Nvidia GPU this driver does not support Intel's iGPU
I don't know if there's a way to skip the environment setup, because I don't have this headset kit now.

Holocala, 2021-12-07:
> Hypereal Pano is a multi camera IR blob tracking based VR headset, this headset start selling from June 2017, Include dual or triple camera, sens controller and the HMD. dual camera provide 240° and triple camera provide 360° tracking.
> The manufacturer also released compatible tools on Windows to work with SteamVR at that time. but after January 2018, the manufacturer was transformation to other industry and no longer to support this headset.
> With tons of updates from SteamVR, the compatible tools are not work with the latest SteamVR(can not detect the key press event in game ).
> Now this VR headset can only work perfectly with old version SteamVR (such as macos_beta, 1.3.23).
> Does anyone interested to support this devices? For necessary, I can giving a triple-Camera headset with the latest client drivers from manufacturer and old version SteamVR macos_beta for reverse engineering.

Hey bro can you give me steam vr 1.3.23 can’t find it everywhere.