OpenHMD/OpenHMD #305: Can't get Rift S working in SteamVR. It's looking for 2560x1440 screen but my main screen has same resolution and SteamVR choses it as correct. When I disconnect main screen SteamVR is not seeing Rift S at all but HMD appearing in xrandr.

thaytan, 2020-06-30:
Can't get Rift S working in SteamVR. It's looking for 2560x1440 screen but my main screen has same resolution and SteamVR choses it as correct. When I disconnect main screen SteamVR is not seeing Rift S at all but HMD appearing in xrandr.
AMD Radeon RX 5700. Arch Linux
vrcompositor log:
```
Tue Jun 30 2020 14:17:26.618298 - Looking for direct display through RandR
Tue Jun 30 2020 14:17:26.618308 -  - Root 0x6cd
Tue Jun 30 2020 14:17:26.618752 -    - Output 0x55 - 0 modes, 0 preferred
Tue Jun 30 2020 14:17:26.618770 -    - Output 0x56 - 13 modes, 1 preferred
Tue Jun 30 2020 14:17:26.618777 -      - Mode 0 0x68 1280x1024 (looking for 2560x1440)
Tue Jun 30 2020 14:17:26.618791 -    - Output 0x57 - 0 modes, 0 preferred
Tue Jun 30 2020 14:17:26.618806 -    - Output 0x58 - 0 modes, 0 preferred
Tue Jun 30 2020 14:17:26.618814 - Tried to find direct display through RandR: (nil)
Tue Jun 30 2020 14:17:26.618821 - Looking for direct display through Vulkan WSI
Tue Jun 30 2020 14:17:26.618829 - Tried to find direct display through Vulkan WSI: (nil)
```
[xrandr log](https://pastebin.com/8rPrTETP)

_Originally posted by @exstrim401 in https://github.com/OpenHMD/OpenHMD/issues/225#issuecomment-651707192_

thaytan, 2020-06-30:
I set up the Rift S to report 2560x1440 as the screen resolution, but actually it's a 1440x2560 rotated display. It might be better if we change OpenHMD to report that rotated configuration and appropriate transformation matrices for a rotated display. I'm not 100% sure that the SteamVR-OpenHMD wrapper can handle that yet though. @ChristophHaag ?

thaytan, 2020-06-30:
What I've been doing for testing so far is to remove the non-desktop property from the display and rotate it with xrandr, then move the SteamVR window onto the display manually.

I also saw there's SteamVR API for specifying EDID VID and PID - we might be able to help SteamVR identify the correct display that way

ChristophHaag, 2020-06-30:
This issue probably belongs rather in STeamVR-OpenHMD.

I have not really played with direct mode on rotated displays but I do have a WVR2 so I can actually test now.

I don't know how to tell SteamVR that a display is rotated, but I would assume it's one of those props you have to set. SteamVR-OpenHMD currently does rotation via the distortion function, which might be incompatible with how SteamVR does rotation "properly".

thaytan, 2020-06-30:
There's probably work to do in both layers

CoreChg, 2020-11-24:
Seems that with the Rift S the display isn't detected at first, but then powers on when a program is using it.
When I boot my system up and start xinit, xrandr can't detect the Rift S in either direct or extended mode.

However, once I start the openhmd-opengl-example program from the build folder, the display powers on and xrandr can detect the Rift S.

I don't know if this helps here or may be related to the issue, but I figured I'd try to add to the conversation.

CoreChg, 2020-11-24:
Before running:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
DisplayPort-0 disconnected primary (normal left inverted right x axis y axis)
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 59 
                supported: 59
        non-desktop: 0 
                range: (0, 1)
```

During program run:

```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
DisplayPort-0 disconnected primary (normal left inverted right x axis y axis)
        EDID: 
                00ffffffffffff003ed2120000000000
                011d0104a50000000e00000000000000
                00000000000001010101010101010101
                01010101010145dfa0385100eea5d20c
                f10c000000000010000000ff0031574d
                47483532304a4a393237000000fc0052
                69667420530a20202020202000000010
                00000000000000000000000000000025
        GAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        DEGAMMA_LUT_SIZE: 4096 
                range: (0, -1)
        GAMMA_LUT: 0 
                range: (0, 65535)
        CTM: 0 
        DEGAMMA_LUT: 0 
                range: (0, 65535)
        TearFree: auto 
                supported: off, on, auto
        vrr_capable: 0 
                range: (0, 1)
        max bpc: 8 
                range: (8, 16)
        underscan vborder: 0 
                range: (0, 128)
        underscan hborder: 0 
                range: (0, 128)
        underscan: off 
                supported: off, on, auto
        scaling mode: None 
                supported: None, Full, Center, Full aspect
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 59 
                supported: 59
        non-desktop: 1 
                range: (0, 1)
   1440x2560     80.00 +
   1280x1024     80.00  
   1440x900      80.00  
   1280x800      80.00  
   1280x720      80.00  
   1024x768      80.00  
   800x600       80.00  
   640x480       80.00
```

Stemt, 2021-03-30:
I'm having the same (or similiar?) issue. Along with SteamVR I also tried running the opengl example. Both generated the proper image to be displayed for the rift s with proper tracking but only displayed it in a window and not on the rift. I also use an rx 5700 so maybe its related to that?

sifiro, 2021-03-31:
> I'm having the same (or similiar?) issue. Along with SteamVR I also tried running the opengl example. Both generated the proper image to be displayed for the rift s with proper tracking but only displayed it in a window and not on the rift. I also use an rx 5700 so maybe its related to that?

Hi, Some notes, OpenHMD doesn’t support direct mode, so If you *only* want use OpenHMD, you must enable your Rift S as an [Additional Monitor](https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode)
But this is not recommendable, Because monado has support for Direct Mode and has compatibility with OpenHMD Devices (but lacks support for controllers).

So, the ideal use case is use monado with their SteamVR Driver as compositor and use OpenHMD for Driver handling.

TheOnlyJoey, 2021-03-31:
> So, the ideal use case is use monado (with their SteamVR Driver because by default only supports OpenXR Programs/Games) as compositor and use OpenHMD for Driver handling.

Or use SteamVR-OpenHMD (https://github.com/ChristophHaag/SteamVR-OpenHMD) To get direct mode and full controller support with OpenHMD drivers.

sifiro, 2021-03-31:
> Or use SteamVR-OpenHMD (https://github.com/ChristophHaag/SteamVR-OpenHMD) To get direct mode and full controller support with OpenHMD drivers.

Wait, when was enabled the DirectMode in SteamVR-OpenHMD?
Last time, I tried did not work from any form. Maybe It’s because rotated display, but that It’s not handled by the OpenHMD driver so doesn’t show anything. (And I tried several times, I only could make works with extended mode)