OpenHMD/OpenHMD #306: Samsung Oddysey+; USB detected, displays not initializing.

yackaro, 2020-07-01:
Similar issue to #295, but on a different headset.
Continuing off of the last comment there,
>Sounds good. If you can provide a PCAP of that device while Windows powers it up, I can >take a look and see if a similar patch can be used. 
>
>_Originally posted by @kflansburg in https://github.com/OpenHMD/OpenHMD/issues/295#issuecomment-652499013_

I tried to get a USB capture in Wireshark, but I am wholly unfamiliar with how to filter down to the USB device I want, and online documentation of WireShark and USBPCap seems to either expect windows to be in a VM within Linux, or is too vague for me to be able to follow. BUT I did manage to figure out how to filter it down to the headset, and was able to save it to a capture file that I have uploaded to [here](https://drive.google.com/file/d/1Vwy8ibQna7HFMC3K0ni1Jd4NUGnTLz0M/view?usp=sharing)
As stated before, I'm unfamiliar with Wireshark, so if there's a better way to filter out unrelated stuff or a better file type to save it as, that would be great to know.

kflansburg, 2020-07-01:
> Originally posted by @kflansburg in #295 (comment)

It’s unfortunately pretty tricky, I usually try to plug it into a port so that it is on its own bus (as seen in lsusb -vt) and then look at the device descriptions in the pcap to identify the address of the device, then apply a wire shark filter for that address as either source or destination. Finally export the filter packets.

I will take a look at the pcap when I get a chance and see if I can figure anything out.

kflansburg, 2020-07-02:
Ok I see two Samsung devices (VendorID: 0x04E8)

ProductID 1: 0x7312
ProductID 2: 0x7084

The second one appears to send a similar HID message as the HP, so it's probably the easiest one to try. I can write the patch,  but @yackaro can you tell me what is printed after the "[II] Model name:" line when you try to run OpenHMD? I need to be able to identify the device.

yackaro, 2020-07-02:
Sorry about the late response, but is the output you were looking for something like running openhmd_simple_example?
I don't know if this is what you wanted, but I'm going to give you the full output dump.

> OpenHMD version: 0.3.0
>num devices: 5
>
>device 0
>  vendor:  Microsoft
>  product: HoloLens Sensors
>  path:    0
>  class:   HMD
>  flags:   04
>    null device:         no
>    rotational tracking: yes
>    positional tracking: no
>    left controller:     no
>    right controller:    no
>
>device 1
>  vendor:  OpenHMD
>  product: External Device
>  path:    (none)
>  class:   HMD
>  flags:   06
>    null device:         no
>    rotational tracking: yes
>    positional tracking: yes
>    left controller:     no
>    right controller:    no
>
>device 2
>  vendor:  OpenHMD
>  product: HMD Null Device
>  path:    (none)
>  class:   HMD
>  flags:   05
>    null device:         yes
>    rotational tracking: yes
>    positional tracking: no
>    left controller:     no
>    right controller:    no
>
>device 3
>  vendor:  OpenHMD
>  product: Left Controller Null Device
>  path:    (none)
>  class:   Controller
>  flags:   0f
>    null device:         yes
>    rotational tracking: yes
>    positional tracking: yes
>    left controller:     yes
>    right controller:    no
>
>device 4
>  vendor:  OpenHMD
>  product: Right Controller Null Device
>  path:    (none)
>  class:   Controller
>  flags:   17
>    null device:         yes
>    rotational tracking: yes
>    positional tracking: yes
>    left controller:     no
>    right controller:    yes
>
>opening device: 0
>[II] 045e:0659 0004:0004:02
>
>[II] opening
>
>[EE] Could not open device with index: 0, check device permissions?
>failed to open device: Could not open device with index: 0, check device permissions?

I think my udev rules for it got wiped, but when I try to run it as root/superuser, it outputs tracking data so fast it wipes the command history, still with no display.

kflansburg, 2020-07-02:
Hmmm, can you redirect the output to a file? Or edit the simple example to
run for fewer iterations?

yackaro, 2020-07-02:
After five attempts, I managed to kill it before it wiped the history. The output seems to be about the same though, with a few added lines at the bottom before it goes into tracking output. To shorten your searching time, I'm only going to post the output that changed.

> opening device: 0
>[II] 045e:0659 0004:0004:02
>
>[II] opening
>
>[EE] Failed to issue command 08: 17 00 01
>
>[EE] Could not read config from the firmware
>
>resolution:              1 x 0
>hsize:                   0.103812 
>[EE] unknown message type: 23
>vsize:                   0.051905 
>lens separation:         0.063000 
>lens vcenter:            0.025953 
>left eye fov:            1.658063 
>right eye fov:           1.658063 
>left eye aspect:         1.000000 
>right eye aspect:        1.000000 
>distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
>control count:           0 
>controls:

kflansburg, 2020-07-02:
Hmmm, it seems that OpenHMD wasn't able to download the device config from
the device. I have found that to be an intermittent problem, so if you just
run it a few times it may eventually work.

If you change line 137 of examples/simple/simple.c from "for(int i = 0; i <
10000; i++){" to "for(int i = 0; i < 1; i++){", (notice the 10,000 changed
to 1) and recompile, that should stop it from printing out the readings and
make it easier to run multiple times.

yackaro, 2020-07-03:
I got it to output the config, but it seems to be a hit-or-miss, with mostly misses, as it took ten tries to get it to output without erroring out on reading the config. I've got the new output here. I accidentally applied your changes to line 130 instead of 137, but that seems to have resulted in the desired, if not intended, effect, just erroring or faulting out at the end, depending on whether it was able to grab config or not.
>opening device: 0
>[II] 045e:0659 0004:0006:02
>
>[II] opening
>
>[II] Read 10217-byte config data
>
>[II] Model name: Samsung Windows Mixed Reality 800ZBA
>
>[EE] Found display height 1600
>
>[EE] Found display width 2880
>
>resolution:              2880 x 1600
>hsize:                   0.103812 
>vsize:                   0.051905 
>lens separation:         0.063000 
>lens vcenter:            0.025953 
>left eye fov:            1.658063 
>right eye fov:           1.658063 
>left eye aspect:         1.000000 
>right eye aspect:        1.000000 
>distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
>control count:           0 
>Bus error

kflansburg, 2020-07-03:
Perfect! Thanks!

kflansburg, 2020-07-03:
Ok, here is my branch:

https://github.com/kflansburg/OpenHMD/tree/odyssey

Make sure you check out the odyssey branch and recompile. 

You might want to modify the example again so you can see the log output. You should see it detect and initialize the Odyssey. Let me know if that activates the device.

yackaro, 2020-07-03:
It doesn't seem to initialize, but I may have potentially done something wrong. Will keep trying though.

yackaro, 2020-07-03:
I suspect that the failure is related to the config more often than not failing to be read. I can't seem to get an instance where it will read it.

kflansburg, 2020-07-03:
Yes, you will need to get it to read the config. I'm not sure what causes
that. I usually have to try 10+ times as well.

yackaro, 2020-07-03:
I got a run to detect the config, but it didn't initialize, and it gave me some new errors during loading the config.

> opening device: 0
>[II] 045e:0659 0004:0008:02
>
>[II] opening
>
>[EE] Unrecognized Command
[EE] Unrecognized Command: 23
[EE] Unrecognized Command: 0
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 94
[EE] Unrecognized Command: 4
[EE] Unrecognized Command: 93
[EE] Unrecognized Command: 6
[EE] Unrecognized Command
[EE] Unrecognized Command: 23
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 1
[EE] Unrecognized Command: 94
[EE] Unrecognized Command: 4
[EE] Unrecognized Command: 93
[EE] Unrecognized Command: 6
>[II] Read 10217-byte config data
>
>[II] Model name: Samsung Windows Mixed Reality 800ZBA

kflansburg, 2020-07-03:
Hmmm, nothing after the model name? That is where it should have initialized. I would ignore those Unrecognized Command errors for now, I get them too occasionally.

yackaro, 2020-07-03:
everything after the model name runs the same as before, with resolution and distortion checks, followed by gyro tracking output.

kflansburg, 2020-07-03:
Hmmmm, are you sure you checked out the branch? Maybe it didn't detect it somehow.

yackaro, 2020-07-03:
That was one of the first things I checked, I tried to check it out again, and git said I was already on it.

kflansburg, 2020-07-03:
Ok, I've updated that branch to skip the check and just do the initialization. Can you pull it, recompile, and try again?

yackaro, 2020-07-03:
Just pulled and recompiled, still nothing. I'll post the relevant output anyway, though.

> opening device: 0
[II] 045e:0659 0004:0008:02
>
>[II] opening
>
>[II] Read 10217-byte config data
>
>[II] Model name: Samsung Windows Mixed Reality 800ZBA
>
>[EE] Found display height 1600
>
>[EE] Found display width 2880
>
>[II] Detected Samsung Odyssey
[II] Initializing Odyssey.
resolution:              2880 x 1600
hsize:                   0.103812 
vsize:                   0.051905 
[EE] unknown message type: 2
lens separation:         0.063000 
lens vcenter:            0.025953 
left eye fov:            1.658063 
right eye fov:           1.658063 
left eye aspect:         1.000000 
right eye aspect:        1.000000 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:

kflansburg, 2020-07-03:
Hmmm, ok, this may not have worked. The next step is probably to try the sequence seen on the other Samsung device. I can take a crack at that next week.

yackaro, 2020-07-06:
Sorry about the late reply, but keep me posted!
The more headsets that can work with OpenHMD IMHO the better.
Hopefully I-O positional tracking can get figured out as well, that would be a huge boon for not only the Rift S, where AFAIK there's someone trying to work on a tracking algorithm to work with it, but all of these WMR headsets as well as the upcoming WMR G2 headsets, right now from HP but pretty much guaranteed to also be from at least Samsung, if not other OEMs as well like Lenovo and Acer; Acer already came close to releasing another new WMR headset, so I'm sure they have something in the works based around WMR G2.