OpenHMD/OpenHMD #307: Samsung Odyssey+ screen stays dark while IMU and rendering works.

tuckjohn, 2020-07-31:
I am trying to develop a 3D VR application using OpenHMD, but I have not been able to get the OpenGL Example to work with my Samsung Odyssey+ on Ubuntu 20.04

The simple example works fine, and prints IMU information to the terminal.
<details>
  <summary>Simple Example Console Output</summary>
  
```
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 045e:0659 0004:0003:02

[II] opening

[II] Read 10206-byte config data

[II] Model name: Samsung Windows Mixed Reality 800ZBA

[EE] Found display height 1600

[EE] Found display width 2880

resolution:              2880 x 1600
hsize:                   0.103812 
vsize:                   0.051905 
lens separation:         0.063000 
lens vcenter:            0.025953 
left eye fov:            1.658063 
right eye fov:           1.658063 
left eye aspect:         1.000000 
right eye aspect:        1.000000 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:                

rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 

[EE] unknown message type: 2
rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000 

[EE] unknown message type: 23
[EE] unknown message type: 23
rotation quat:           0.000000 0.000000 0.000000 1.000000 
position vec:            0.000000 0.000000 0.000000

rotation quat:           -0.000017 0.000062 -0.000057 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.000036 0.000131 -0.000120 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.000066 0.000251 -0.000214 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.000082 0.000315 -0.000276 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.000099 0.000380 -0.000337 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104717 0.000242 0.698941 0.707471 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104696 0.000242 0.698891 0.707524 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104648 0.000261 0.698822 0.707599 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104587 0.000295 0.698784 0.707646 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104467 0.000385 0.698734 0.707713 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104400 0.000434 0.698700 0.707756 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104340 0.000473 0.698651 0.707814 
position vec:            0.000000 0.000000 0.000000 
[Continues printing information]
```

</details>

The OpenGL example doesn't start most of the time(But this seems like normal behavior, and is not the problem I hope to fix here), but when it does it renders the scene with colored shapes on my monitor, but the displays in the headset stay dark. It seems to be using the sensors from the headset correctly (eg. I turn the headset left the view moves left). Console output from both are below.
<details>
  <summary>OpenGL Example Bad Console Output</summary>

I get this maybe 50% the time
 ```
[II] 045e:0659 0004:0003:02

[II] opening

[EE] Failed to issue command 08: 17 00 01

[EE] Could not read config from the firmware

[EE] unknown message type: 23
[EE] unknown message type: 2
SDL_CreateWindow failed
```

</details>
<details>
  <summary>OpenGL Example Bad 2(Doesn't create render window) Console Output</summary>

I get this maybe 30% the time
```
[II] 045e:0659 0004:0003:02

[II] opening

[EE] Failed to issue command 08: 17 00 01

[EE] Could not read config from the firmware

[EE] unknown message type: 23
[EE] unknown message type: 2
OpenGL Renderer: GeForce GTX 1050 Ti/PCIe/SSE2
OpenGL Vendor: NVIDIA Corporation
OpenGL Version: 4.6.0 NVIDIA 435.21
failed to create fbo 8cd6
failed to create fbo 8cd6
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23

[Continues until Ctrl+C]
```

</details>
<details>
  <summary>OpenGL Example "Good" Console Output</summary>

  I get this maybe 20% the time
  
```
[II] 045e:0659 0004:0003:02

[II] opening

[II] Read 10206-byte config data

[II] Model name: Samsung Windows Mixed Reality 800ZBA

[EE] Found display height 1600

[EE] Found display width 2880

[EE] unknown message type: 2
[EE] unknown message type: 23
[EE] unknown message type: 23
OpenGL Renderer: GeForce GTX 1050 Ti/PCIe/SSE2
OpenGL Vendor: NVIDIA Corporation
OpenGL Version: 4.6.0 NVIDIA 435.21
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23
[EE] unknown message type: 23

[Continues until Ctrl+C]
```

</details>
I have looked at issues /issues/276, /issues/295, and /issues/306, and they seem to mirror my problem exactly, but I was not able to find a fix.

If I run `xrandr`, It does not seem to see the display in the headset (Which is plugged into the HDMI plug in my graphics card). This confused me, as it is able to see the resolution of it when running the simple sample (Unless I am misunderstanding something).

<details>
  <summary>xrandr Output</summary>

```
$ xrandr
Screen 0: minimum 8 x 8, current 2960 x 1050, maximum 32767 x 32767
DVI-D-0 connected 1280x1024+1680+0 (normal left inverted right x axis y axis) 380mm x 300mm
   1280x1024     60.02 +  75.02* 
   1280x960      60.00  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   800x600       75.00    72.19    60.32    56.25  
   640x480       75.00    72.81    59.94  
HDMI-0 disconnected (normal left inverted right x axis y axis)
DP-0 disconnected (normal left inverted right x axis y axis)
DP-1 connected primary 1680x1050+0+0 (normal left inverted right x axis y axis) 526mm x 296mm
   1680x1050     59.88*+
   1440x900      59.89  
   1280x1024     75.02    72.00    70.00    60.02  
   1280x960      60.00  
   1152x864      75.00  
   1024x768      75.03    72.00    70.07    60.00  
   800x600       75.00    72.19    60.32    56.25  
   640x480       75.00    72.81    59.94   
```
</details>

How can I get the displays in the headset to work? I have only be using Linux for a few months now, so I am pretty new to a lot of this. Let me know if there is anything else I can do to get more information on the problem.

pH5, 2020-08-01:
> If I run `xrandr`, It does not seem to see the display in the headset (Which is plugged into the HDMI plug in my graphics card). This confused me, as it is able to see the resolution of it when running the simple sample (Unless I am misunderstanding something).

The native panel resolution reported by the simple example is read from a JSON configuration block via USB, so this is not a sign that the HDMI connection is active.

tuckjohn, 2020-08-02:
Shows how much I know... 

Does the headset need some sort of data packet to turn on the displays, that only Windows sends it?

thaytan, 2022-02-08:
The Monado WMR driver has support for the Samsung Odyssey+