OpenHMD/OpenHMD #308: Starting work on positional tracking (for WMR currently

herosilas12, 2020-08-03:
I made a comment on a different thread but it was an oculus thread so that is my bad
anyways

I need to know if openHMD has full camera access for WMR headsets

if it does, how do I access said cameras

then, I have only worked in python. is there someway to integrate it or should I work on learning C

thanks

ChristophHaag, 2020-08-03:
Currently OpenHMD has no camera driver and so far has preferred not depending on libusb directly. I assume some infrastructure to read camera streams in OpenHMD will come with the Oculus Rift CV1 tracking though.

For the WMR cameras there is one important distinction: HMDs with old firmware expose cameras as standard UVC cameras, with new firmware they are exposed via proprietary protocol. The new protocol has been reverse engineered and there is a prototype implementation in pull request #208. Currently the only way to update the firmware is to connect the HMD to a windows machine, probably requires starting the MR portal application.

I don't think OpenHMD will accept python into the main codebase, but writing a first prototype in python is certainly a possibility.

How would you plan to work positional tracking? Markerless inside out tracking is quite a different beast than tracking LEDs on a device, the most feasible way to do something here is likely to get already existing SLAM algorithms working. For the possibility of upstreaming such a driver into OpenHMD it would be good not to reuse any GPL code.

Examples of MIT licensed SLAM frameworks/libraries are maplab, openvslam or kimera, though as far as we know none of them have been tuned for "real time performance" required by VR.

thaytan, 2020-08-04:
A full SLAM implementation is not necessary for doing HMD tracking. The VIO (Visual Inertial Odometry) subset of a SLAM implementation should be sufficient. That said - if there's an existing SLAM implementation that is fast enough and easy to integrate, that would be the easiest path - and would allow cool features like generating actual maps of people's play areas for use in the VR world!

the Rift CV1 positional branch has an implementation of a UVC camera driver with the required tweaks to interpret CV1 data properly. I'm not sure how much of that is useful WMR - maybe for the 'older firmware' models. I'm planning to rework that implementation at some point, so keep that in mind when copying any code. I want to make it support separating the capture and processing into separate threads - so that analysis never trips up capture.

thaytan, 2020-08-04:
By the way, I'm happy to provide support for this but I don't have any WMR headsets at the moment so can't join in directly.

herosilas12, 2020-08-06:
The way I figured I would get it to work is place digital markers arounds the room and calculate the distance from those with some rotation data and get position that way
I haven't worked out how I am gonna do the controllers but it might become clear at some point along the way
and to be clear, only MIT licenses?

herosilas12, 2020-08-06:
I am almost done with the renderer I am gonna use for testing
I just need to install the bindings for python
also do you think someone could make an openHMD discord server
I have tried asking questions on the IRC but whenever I close the window and reopen it it only lists new messages
could I make the discord and transfer ownership to one of you?

TheOnlyJoey, 2020-08-06:
To not split up the development discussions, we prefer to stay on IRC.
Most people have a bouncer (something like ZNC or Matrix) to keep persistence on the server, some just always have their IRC window open.

thaytan, 2020-08-07:
I already have a discord window open for other things, so I don't personally have a problem with the idea - but as Joey says everyone is on IRC right now, and I don't know whether a discord would get traction with the few people you want to be able to talk to.
Something like irccloud might work for you to maintain some persistence, or Matrix as Joey suggested.

herosilas12, 2020-08-07:
I wound up setting up matrix
the UI is really nice
anyway I have some issues
1. I can't get libUVC to compile which I need for image merging 
2. I can't get the python-openHMD binding to compile due to various errors 
[Errors.txt](https://github.com/OpenHMD/OpenHMD/files/5041227/Errors.txt)

herosilas12, 2020-08-07:
I do intend to set up a fedora dual boot soon and I already ordered another SSD so that should be here by Saturday

TheOnlyJoey, 2020-08-07:
@herosilas12 Can you please not use OpenHMD issues as a generic chat service? if you want to ask questions please join the IRC, or send emails. Please keep the issues on topic.
Not trying to be demotivating or harsh, but if you can't figure out how to compile certain libraries, I am pretty sure inside out tracking is going to be a bit to much probably....

closed 2021-06-08 by herosilas12