OpenHMD/OpenHMD #310: Rift S driver issue?

thaytan, 2020-08-12:
> But right now I'm debugging some really weird behavior in the Rift S driver, in order to further familiarize myself with OpenHMDs code.

_Originally posted by @ChaiyaBili in https://github.com/OpenHMD/OpenHMD/pull/309#issuecomment-672994788_

Do you have any more details? What sort of behaviour?

ChaiyaBili, 2020-08-12:
Hi,

well I haven't created an issue yet, because I'm rather new to OpenHMD and maybe I'm just using it wrong. Also debugging a  newly discovered software is my way to understand how it works.
So take my comments with a grain of salt.

Right now I'm investigating two things.
1. Opening the devices for the left or right controller doesn't work reliably. When I call  ohmd_list_open_device OpenHMD sometimes notifies me that it couldn't read the controller calibration data. This happens sporadically. Sometimes it just works sometimes it doesn't. When it doesn't work ohmd_list_open_device will still return a valid handle but when I query the IMU via that handle, the returned quaternion will always be 0,0,0,1.

2. When closing the devices afterwards with ohmd_close_device only the closing of the first device will succeed. Closing the remaining devices will raise an error that there isn't an open HMD device in the device list anymore. Also the call to ohmd_ctx_destroy will then throw a SIGABRT sometimes a SIGSEGV after I got that error message.
Sometimes my Rift S would also be unreponsive after ohmd_ctx_destroy failed this way, and I have to unplug/replug it to my USB port.

The pseudocode to reproduce this would be:
`
//Create context
ctx = ohmd_ctx_create();

//Open devices
ohmd_device* hmd = ohmd_list_open_device(ctx, 0); //Works always
ohmd_device* left= ohmd_list_open_device(ctx, 1); //Works sometimes
ohmd_device* right = ohmd_list_open_device(ctx, 2); //Works sometimes

//Do something meaningful with the IMU data until the user ends the program

//Close devices
ohmd_close_device(right);  //This works
ohmd_close_device(left);    //This will raise an error
ohmd_close_device(hmd); //This will raise an error

//Destroy context
ohmd_ctx_destroy(ctx); //This will sometimes throw a SIGABRT sometimes even a SIGSEGV
`

I'm using Gentoo Linux
kernel: 4.19.130-gentoo
platform: x86_64-pc-linux-gnu,

Cheers,
Chaiya

thaytan, 2020-08-12:
I saw a report on twitter without any detail that sometimes the controllers fail to start up on opening that sounds similar. I haven't encountered it - but perhaps I just haven't done enough testing with programs that access all both HMD and controllers simultaneously on the Rift S.

Looking at the code, I see there's a bug (in the Rift CV1 driver too) around not tracking the use_count properly. The idea is that because all reports from the HMD and controllers actually flow through the HMD, the first call to open_device() creates a shared object, and subsequent calls to open the other devices should then just increment the use_count, and close_device() should decrement it.... looks like the shared object is being freed too early (on the first call to close_device()) which would explain the crashes.

As for the the failure to start up - I expect that open_device() will always succeed, but controllers don't start to report their IMU information until they connect to the headset and are online. My reverse engineered startup sequence might fail to wake up the controllers sometimes I found hitting a button on the controller can wake it up, and then it starts sending.

thaytan, 2020-08-12:
This should fix the crashes on close (untested):

```
diff --git a/src/drv_oculus_rift_s/rift-s.c b/src/drv_oculus_rift_s/rift-s.c
index 77cef4b..44ee4e1 100644
--- a/src/drv_oculus_rift_s/rift-s.c
+++ b/src/drv_oculus_rift_s/rift-s.c
@@ -55,8 +55,10 @@ static rift_s_hmd_t *find_hmd(char *hid_path)
        device_list_t* current = rift_hmds;
 
        while (current != NULL) {
-               if (strcmp(current->path, hid_path)==0)
+               if (strcmp(current->path, hid_path)==0) {
+                       current->hmd->use_count++;
                        return current->hmd;
+               }
                current = current->next;
        }
        return NULL;
```

ChaiyaBili, 2020-08-12:
Wow, this was fast :smiley_cat: 

I just tested your patch and indeed close_device and ctx_destroy now work properly.

thaytan, 2020-08-15:
With the crashing fixed, are you still having trouble with the controllers not sending reports sometimes?

ChaiyaBili, 2020-08-17:
Yes at the moment that's the only problem.
I'm currently on vacation and the internet is rather bad here.
That's why I couldn't answer earlier.

thaytan, 2020-08-18:
Thanks for the update!

closed 2021-05-02 by thaytan