OpenHMD/OpenHMD #312: Monado's 6DOF PSVR?

poVoq, 2020-09-07:
According to the Monado website ( https://monado.freedesktop.org/ )and a presentation from earlier this year, they got PSVR somewhat working with 6DOF?

Is that something that can be used with OpenHMD or somehow backported?

TheOnlyJoey, 2020-09-07:
We are aware of their work, some of their team are OpenHMD contributors :)
There might be parts that are interesting for us, but they do rely on some external dependencies and processes that are probably not compatible with OpenHMD.

Our idea currently is to first get the DK2/CV1 tracking done (going strong by @thaytan at https://github.com/thaytan/OpenHMD/tree/rift-correspondence-search), and try to work from there to do other types of outside-in tracking for things like the PSVR.

bevr-nz, 2020-09-07:
I wrote the PSVR tracking stuff we have in Monado. 

The tracker itself most likely uses fewer dependencies and external code than it could, but much of the infrastructure around calibration and video input/filtering is unlikely to meet OpenHMDs requirements, and taken as a whole, it is probably larger in terms of LOC that the entirety of OpenHMD anyway.

I'll avoid getting into the weeds here, but suffice to say I have followed the OpenHMD project for a long time, and despite my involvement in all things OpenXR - do appreciate the light and simple API that OHMD represents - but I am not optimistic that any extant 6DOF tracker could be merged into OpenHMD in the short term given the constraints in place, without reliance on magical thinking. 

I'd love to be shown to be wrong.

TheOnlyJoey, 2021-01-24:
After the DK2/CV1 branch gets upstreamed, I will start working on getting PSVR tracking done, already have some older test projects that i can revive to speed things up. (Tracking update can be found on https://www.youtube.com/watch?v=3AdmS3vy7ZE).
Already have a bunch of patches ready from last year to update the PSVR to re-enable in release (Windows support).
Closing for now.

closed 2021-01-24 by TheOnlyJoey