OpenHMD/OpenHMD #314: MacOS compilation LG-R100

beckmx, 2020-10-01:
Hello guys, I ran a compilation for the LGR100 against the recommended branch, actually as of today with the latest version of catalina 10.15.7, I didn't have any problems, only if I select the OpenGL example, it throws some problems about API deprecation, but the driver seems to compile. Now, after the `make` and `make install` I ran the simple script but the only data I see is this:

```
OpenHMD version: 0.3.0
num devices: 4

device 0
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
resolution:              1280 x 800
hsize:                   0.149760
vsize:                   0.093600
lens separation:         0.063500
lens vcenter:            0.046800
left eye fov:            2.190639
right eye fov:           2.190639
left eye aspect:         0.800000
right eye aspect:        0.800000
[EE] invalid type given to getf (18)
distortion k:            0.800000 0.000000 0.000000 0.000000 0.000000 0.000000
control count:           0
controls:

rotation quat:           0.000000 0.000000 0.000000 1.000000
position vec:            0.000000 0.000000 0.000000

rotation quat:           0.000000 0.000000 0.000000 1.000000
position vec:            0.000000 0.000000 0.000000

rotation quat:           0.000000 0.000000 0.000000 1.000000
position vec:            0.000000 0.000000 0.000000

rotation quat:           0.000000 0.000000 0.000000 1.000000
position vec:            0.000000 0.000000 0.000000

```

and continuously repeats the "rotation quat" but doesn't change any of the numbers, I was wondering if any of you have a better idea on what's going on

TheOnlyJoey, 2020-10-02:
By default the simple example opens the first device (device 0) which is External Device. This means that is has not been able to open the device. Are you sure you have the LG connected with the https://github.com/OpenHMD/OpenHMD/tree/LG-R100 branch?

beckmx, 2020-10-02:
Thanx for the reply @TheOnlyJoey I rebuilt everything for you to see my compilation log, in case you see something weird:

```
bash@bash-3344  ~/OpenHMD   LG-R100 ?  make                                               ✔  9797  21:43:23
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       libopenhmd_la-openhmd.lo
  CC       libopenhmd_la-platform-win32.lo
  CC       drv_dummy/libopenhmd_la-dummy.lo
  CC       libopenhmd_la-omath.lo
  CC       libopenhmd_la-platform-posix.lo
  CC       libopenhmd_la-fusion.lo
  CC       libopenhmd_la-shaders.lo
  CC       drv_oculus_rift/libopenhmd_la-rift.lo
  CC       drv_oculus_rift/libopenhmd_la-packet.lo
  CC       drv_htc_vive/libopenhmd_la-vive.lo
  CC       drv_htc_vive/libopenhmd_la-packet.lo
  CC       ext_deps/libopenhmd_la-nxjson.lo
  CC       drv_deepoon/libopenhmd_la-deepoon.lo
  CC       drv_deepoon/libopenhmd_la-packet.lo
  CC       drv_wmr/libopenhmd_la-wmr.lo
  CC       drv_wmr/libopenhmd_la-packet.lo
  CC       drv_psvr/libopenhmd_la-psvr.lo
  CC       drv_psvr/libopenhmd_la-packet.lo
  CC       drv_nolo/libopenhmd_la-nolo.lo
  CC       drv_nolo/libopenhmd_la-packet.lo
  CC       drv_3glasses/libopenhmd_la-xgvr.lo
drv_3glasses/xgvr.c:91:12: warning: unused function '_send_feature_report' [-Wunused-function]
static int _send_feature_report(xgvr_priv* priv, const unsigned char *data, size_t length)
           ^
1 warning generated.
  CC       drv_3glasses/libopenhmd_la-packet.lo
drv_3glasses/packet.c:29:20: warning: unused function 'float_to_byte' [-Wunused-function]
inline static void float_to_byte(const float f, unsigned char *byte)
                   ^
drv_3glasses/packet.c:64:23: warning: unused function 'read_int16' [-Wunused-function]
inline static int16_t read_int16(const unsigned char** buffer)
                      ^
drv_3glasses/packet.c:71:24: warning: unused function 'read_uint32' [-Wunused-function]
inline static uint32_t read_uint32(const unsigned char** buffer)
                       ^
drv_3glasses/packet.c:78:23: warning: unused function 'read_int32' [-Wunused-function]
inline static int32_t read_int32(const unsigned char** buffer)
                      ^
drv_3glasses/packet.c:111:20: warning: unused function '_quat_inv' [-Wunused-function]
static inline void _quat_inv(float *in, float *out)
                   ^
5 warnings generated.
  CC       drv_external/libopenhmd_la-external.lo
  CCLD     libopenhmd.la
copying selected object files to avoid basename conflicts...
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: .libs/libopenhmd.a(libopenhmd_la-platform-win32.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: .libs/libopenhmd.a(libopenhmd_la-platform-win32.o) has no symbols
Making all in tests
Making all in unittests
  CC       main.o
  CC       quat.o
  CC       vec.o
  CC       highlevel.o
  CCLD     unittests
make[3]: Nothing to be done for `all-am'.
Making all in examples
Making all in simple
  CC       simple.o
  CCLD     simple
make[3]: Nothing to be done for `all-am'.
make[2]: Nothing to be done for `all-am'.
bash@bash-3344  ~/OpenHMD   LG-R100 ?  sudo make install                                  ✔  9798  21:43:40
Password:
Making install in src
 /usr/local/bin/gmkdir -p '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /usr/local/bin/ginstall -c   libopenhmd.la '/usr/local/lib'
libtool: install: /usr/local/bin/ginstall -c .libs/libopenhmd.0.dylib /usr/local/lib/libopenhmd.0.dylib
libtool: install: (cd /usr/local/lib && { ln -s -f libopenhmd.0.dylib libopenhmd.dylib || { rm -f libopenhmd.dylib && ln -s libopenhmd.0.dylib libopenhmd.dylib; }; })
libtool: install: /usr/local/bin/ginstall -c .libs/libopenhmd.lai /usr/local/lib/libopenhmd.la
libtool: install: /usr/local/bin/ginstall -c .libs/libopenhmd.a /usr/local/lib/libopenhmd.a
libtool: install: chmod 644 /usr/local/lib/libopenhmd.a
libtool: install: ranlib /usr/local/lib/libopenhmd.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /usr/local/lib/libopenhmd.a(libopenhmd_la-platform-win32.o) has no symbols
 /usr/local/bin/gmkdir -p '/usr/local/include/openhmd'
 /usr/local/bin/ginstall -c -m 644 ../include/openhmd.h '/usr/local/include/openhmd'
Making install in tests
Making install in unittests
 /usr/local/bin/gmkdir -p '/usr/local/bin'
  /bin/sh ../../libtool   --mode=install /usr/local/bin/ginstall -c unittests '/usr/local/bin'
libtool: install: /usr/local/bin/ginstall -c unittests /usr/local/bin/unittests
make[3]: Nothing to be done for `install-data-am'.
make[3]: Nothing to be done for `install-exec-am'.
make[3]: Nothing to be done for `install-data-am'.
Making install in examples
Making install in simple
 /usr/local/bin/gmkdir -p '/usr/local/bin'
  /bin/sh ../../libtool   --mode=install /usr/local/bin/ginstall -c simple '/usr/local/bin'
libtool: install: /usr/local/bin/ginstall -c simple /usr/local/bin/simple
make[3]: Nothing to be done for `install-data-am'.
make[3]: Nothing to be done for `install-exec-am'.
make[3]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
 /usr/local/bin/gmkdir -p '/usr/local/lib/pkgconfig'
 /usr/local/bin/ginstall -c -m 644 pkg-config/openhmd.pc '/usr/local/lib/pkgconfig'
bash@bash-3344  ~/OpenHMD   LG-R100 ? 
```

beckmx, 2020-10-05:
maybe is a problem with the MacOS, I tried to install VMware Fusion, and the device keeps disconnecting every 2 seconds, I know for a fact the glasses work, because I have the lg g5 :(