OpenHMD/OpenHMD #315: PSVR fails at power on function

gb2111, 2020-10-01:
Hi,
I am testing PSVR and it fails when trying [turn the display on](https://github.com/OpenHMD/OpenHMD/blob/f8f8b2dfed4a92aa4c6e5060017390f99b71ec6a/src/drv_psvr/psvr.c#L262).

if I comment out this and also set VR mode code then rest works fine and can read parameters and read the rotation.
Any help is welcome.

TheOnlyJoey, 2020-10-02:
What OS are you running, version? any logs?

gb2111, 2020-10-02:
apologies, i am running windows 10. 

logs from c# test code:
num devices:  1
Device: 0
  Vendor: Sony
  Product: PSVR
  Path: 0
Could not open device with index: 0, check device permissions?

I also tested with trinus app and works without the problems.

But inside c++ code I can see it opened interface 4 and interface 5 and then failed at this specific line.
Thanks for reply and please let me know if any other info is required.

TheOnlyJoey, 2020-10-02:
Duplicate of #186 , closing

closed 2020-10-02 by TheOnlyJoey