OpenHMD/OpenHMD #316: OpenHMD chooses wrong display

dikonov, 2020-11-22:
I am trying to setup a VR desktop based on Safespaces (which uses OpenHMD) and it turned out that all OpenHMD stuff fails to appear on the HMD screen. I got my HMD detected as a second screen using a eGPU box with Nvidia RTX2070 inside (blob driver 450.80.02). The HMD is HTC Vive PRO. 

Both safespaces and openhmd example app run in SBS stereo and react to the HMD motion correctly. The problem is that I cannot get them display that stereo in the HMD! All output always goes to a window on the main display.

1. I can see the plain 2D rendition of my desktop in the HMD
2. I run the openhmd_opengl_example.
3. The HMD screen darkens (it is still on, but shows darkness) and stays dark. At the same time the SBS video stream from the VR app appears in a window on the main display. 
    If I start the VR app form a terminal on the HMD display, the window still appears on the MAIN display. 
    Drugging the VR window to the HMD does not help. The HMD stays dark.
4. I quit the VR app and the HMD either returns to its previous state as a 2D display (IceWM - a plain 2D WM) or stays dark (Cinnamon DE with own GL compositor).

TheOnlyJoey, 2020-11-22:
Closed for being duplicate.
TL;DR: OpenHMD is not responsible for anything screen/application management related since that falls under the task of a compositor. Either use a compositor or enable extended mode (which should be the default option for Safespaces). https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode#using-extended-mode-on-nvidia-linux

closed 2020-11-22 by TheOnlyJoey

dikonov, 2020-11-22:
This link is completely irrelevant because 
1. I have the AllowHMD options set and non-desktop property cleared.  The desktop is already extended to the HMD.
2. This is not a window management issue, because I tried completely DIFFERENT WMs (simple 2D and advanced gl-compositing) and it made no difference to the fact that the VR video will always go to the non HMD desktop even with active HMD display alongside.
3. OpenHMD definitely knows, where the HMD is and tries to control it. It is able to suppress the regular desktop and release the HMD when VR is closed.

Therefore, the bug is real and is not a misconfiguration as suggested by the linked info.

TheOnlyJoey, 2020-11-22:
1: If Extended Mode is working, and the device is tracking correctly, then it works properly.
2-3: OpenHMD does not do _any_ type of management of the rendering side, just implements drivers and communicating through the USB and HID protocol. Things like window management or direct mode rendering is up to a Compositor, which OpenHMD is not.

We do have a universal shader, but thats not a compositor, just a shader implemented in OpenGL that applications can use if they want. A example of a Compositor would be Monado (OpenXR runtime that supports OpenHMD devices) which has a compositor that does direct mode rendering and does not require extended mode. Safespaces does require extended mode since i think it does not include any compositor, but thats something you should check with them.

dikonov, 2020-11-23:
Facts: 
- The VivePron HMD works in both extended and direct modes
- The devs of both projects OpenHMD and Safespaces swear that their code works well. 
- The actual behavior is that I see black field in HMD and a preview(?) non-composited VR window. Openhmd_opengl_example and Safespaces do the same under different WMs.

Theory:  **The glsl shader is broken ?**.
Let us take the openhmd_opengl_example, which certainly uses the universal shader. I assume that the dark field is a product of the broken shader. How can I debug the glsl shader? Is there a log somewhere? What version of glsl libs is tested and known to work?  

I have glslang 8.13.3743 installed from ALTLinux Sisyphus distribution repositories. My OpenHMD package is git-master built by myself.