OpenHMD/OpenHMD #317: Pimax Vision 8kX Support - Important HMD! PCAP files included.

mirage335, 2020-11-30:
Please find PCAP and related files attached, with data from connecting both the Pimax Vision 8kX and Pimax 5k+ .
[Pimax_PCAP_Captures.zip](https://github.com/OpenHMD/OpenHMD/files/5615633/Pimax_PCAP_Captures.zip)

Please consider that the Pimax Vision 8kX has near typical human vision characteristics - 20/20 Snellen test readability, wide FOV, and aftermarket comfort accessibility. Also, the Pimax Vision 8kX uses the same Valve Lighthouse based tracking/controllers widely used by other headsets.

Meanwhile, SteamVR and MSW are such unstable and massive obstructions to software interoperability that more developer time has been spent just getting desktop overlays to work, than would have been necessary to bring OpenHMD support to most VR headsets. Arguably, professional or long-term applications of VR, require multitasking, transparent 2D framebuffer overlays, multi-headset and CAVE outputs, which is at a severe conflict of interest with such commercial proprietary middleware.

I feel it is now as imperative to bring such full VR capability to FOSS, running on Linux, as historically, the Linux kernel and open-source web browsers have proven.

duplexsystem, 2022-09-28:
Any news on this? I'm willing to test or provide another capture of the 5k+ if needed

RoxyBoxxy, 2023-02-07:
Also wanting this, on the pimax artisan

SpookySkeletons, 2023-10-02:
For anyone still in possession of a pi, looks like it may work on all P2 HMDs
https://gitlab.freedesktop.org/haagch/monado/-/commits/5k+

mirage335, 2023-10-04:
Oh. That looks interesting.

We also really need 8kX, and now Crystal/Reality compatibility though. The higher resolution of those headsets is reaching an important point of diminishing returns in usability.

TrulySc, 2023-12-16:
Probably a lost cause, but I'd like to pledge help with testing/dumping anything needed for the 8kx specifically.