OpenHMD/OpenHMD #318: [Feature Request] Vive Cosmos and Cosmos Elite support

khionu, 2020-12-05:
Support for these headsets broke because they don't use Lighthouse technology anymore. SteamVR doesn't even have support on Linux.

DieserMerlin, 2021-03-24:
The Cosmos Elite still uses Lighthouse and everything under Linux is working fine except for the displays on the HMD. Tracking works well (in the SteamVR preview window).

This is so frustrating

achiestdragon, 2022-01-22:
any progress? , it would help to have support for Vive Cosmos

khionu, 2022-01-23:
> since the [ads] do say PC Mac Linux

What ads?

DieserMerlin, 2022-01-23:
There are some YouTube videos and shop pages advertising itHolen Sie sich Outlook für Android

achiestdragon, 2022-01-23:
add's like ,,   https://www.youtube.com/watch?v=MGMJF_NYBuY  ,, or https://ne-np.facebook.com/worldonlineshopstore/videos/2992651167662546/    to name but a few

khionu, 2022-01-24:
That's not an official ad, and it's a mistake.

achiestdragon, 2022-01-24:
eather way  support for it in linux would be good
from what i can tell , reading forums etc , as it stands (ok using steamvr in linux) others have got as far as :-  
  getting the controlers to work 
  getting the hud tracking to work 
  recognising the usb devices

but all seem to fail on one main point, that they cant get the hud display to turn on 
given its just a screen ,with reports of xrandr showng the hud display but its disabled (unable to enable it )

theres a lot of posts regarding this issue although its not clear if its the cosmos spesificaly
about it beeing disabled as a x desktop screen by the os , some reporting it turning on during grub boot but off agian as it boots to x , i cant get xrandr to show it yet 
 
but it dosent sound to me like its far from beeing supported from the hardware side of things
and my actualy find i can get it to work to some extent  

atm im getting the usb devices showing  but stuck there atm  
waiting for bluetooth dongle (in post) for my pc to try at getting the controlers to work
before having another go at getting it runnng  , 

as for OpenHMD , i got as far as this post and not tried it as yet given no support but 
if i may ask , how close is OpenHMD on support for it atm ? ,
and would it be worth trying OpenHMD in its current state  ?

Redhacker1, 2022-08-15:
Necroing this, it would be nice to get the vive cosmos elite working in Linux, from what I am understanding, it sounds like it is largely there with everything but the screen seeming to work. It is a nice headset with index controllers and SteamVR tracking (inbuilt headset tracking is awful). Kinda makes me wish I knew where to even start with a problem like this and how to program in C lol.

NewtSoup, 2025-02-26:
> I've been considering the headset, I'm just upset it may still not have compatibility with Linux.

It's not that it's not compatible, it's that VR for Linux in general h as been broken for 2 years.  it worked fine at 1.26 but at 1.27 and greater it has had severe performance issues.   There is a custom version of OpenVR.dll available that uses DLSS and AMD FSR by default for games and is configurable by a txt file which helps a number of games. 

For me  Skyrim VR runs great but Fallout 4 VR is unplayable.   Subnautica in VR mode which was THE ONLY way to play at 1.26 is now broken to the point of reprojectile-vomit 

if you want native support in VR in Linux then the Index is still the only way to do it.

DieserMerlin, 2025-02-26:
There's still no display driver unfortunately