OpenHMD/OpenHMD #319: openhmd_simple_example gives weird output to PSVR but openhmd_opengl_example works fine

odecif, 2020-12-30:
Hello first time poster in this repo. Thanks for an awesome product!

I am running the latest git master of openhmd and I intend to use the openhmd-steamvr package as well, but for now I have problems with the opehmd_**simple**_example.

The openhmd_**opengl**_example works great in the PSVR headset and I can see the cubes all around me, but when I try the simple example one eye focuses on one side and the other eye focuses on the other side and the monitor has a weird focus (see attachment).

I have recently updated the headset from my PS4 to the latest version.

Here is my output of `inxi -SCGm`
```
System:    Host: odecif-y700 Kernel: 5.4.80-2-MANJARO x86_64 bits: 64 Console: tty 2 Distro: Manjaro Linux 
Memory:    RAM: total: 31.20 GiB used: 4.02 GiB (12.9%) 
           Array-1: capacity: 32 GiB slots: 2 EC: None 
           Device-1: ChannelA-DIMM0 size: 16 GiB speed: 2133 MT/s 
           Device-2: ChannelA-DIMM1 size: No Module Installed 
           Device-3: ChannelB-DIMM0 size: 16 GiB speed: 2133 MT/s 
           Device-4: ChannelB-DIMM1 size: No Module Installed 
CPU:       Info: Quad Core model: Intel Core i5-6300HQ bits: 64 type: MCP L2 cache: 6144 KiB 
           Speed: 800 MHz min/max: 800/3200 MHz Core speeds (MHz): 1: 819 2: 803 3: 803 4: 835 
Graphics:  Device-1: Intel HD Graphics 530 driver: i915 v: kernel 
           Device-2: NVIDIA GM107M [GeForce GTX 960M] driver: nvidia v: 450.80.02 
           Device-3: Intel Intel RealSense 3D Camera (Front F200) type: USB driver: uvcvideo 
           Display: server: X.Org 1.20.10 driver: modesetting,nvidia resolution: 1: 1920x1080~60Hz 2: 1920x1080 
           OpenGL: renderer: GeForce GTX 960M/PCIe/SSE2 v: 4.6.0 NVIDIA 450.80.02 
```
I am using the Nvidia-card when trying the examples.

Output of `openhmd_opengl_example`
```
[odecif@odecif-y700 ~]$ openhmd_opengl_example 
[II] 054c:09af 0001:0013:04
[II] 	opening '0001:0013:04'
[II] 054c:09af 0001:0013:04
[II] 054c:09af 0001:0013:05
[II] 	opening '0001:0013:05'
OpenGL Renderer: GeForce GTX 960M/PCIe/SSE2
OpenGL Vendor: NVIDIA Corporation
OpenGL Version: 4.6.0 NVIDIA 450.80.02
```
Output of `openhmd_simple_example`
```
[odecif@odecif-y700 ~]$ openhmd_simple_example 
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Sony
  product: PSVR
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 054c:09af 0001:0013:04
[II] 	opening '0001:0013:04'
[II] 054c:09af 0001:0013:04
[II] 054c:09af 0001:0013:05
[II] 	opening '0001:0013:05'
resolution:              1920 x 1080
hsize:                   0.126000 
vsize:                   0.071000 
lens separation:         0.063100 
lens vcenter:            0.039490 
left eye fov:            1.807638 
right eye fov:           1.807638 
left eye aspect:         0.888889 
right eye aspect:        0.888889 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           3 
controls:                volume-up (digital), volume-down (digital), mic-mute (digital)

rotation quat:           0.000002 -0.000152 -0.000022 1.000000 
position vec:            0.000000 0.000000 0.000000 
controls state:          0.000000 0.000000 0.000000 
rotation quat:           0.000028 -0.000134 -0.000025 1.000000 
position vec:            0.000000 0.000000 0.000000 
controls state:          0.000000 0.000000 0.000000 
rotation quat:           -0.000012 -0.000142 -0.000022 1.000000 
position vec:            0.000000 0.000000 0.000000 
controls state:          0.000000 0.000000 0.000000 
......
```
I am using Budgie Desktop Manager.

If I quit the simple example the monitor stays the same (weird) until I either reset the PSVR-box or start the opengl example.

The package is installed from openhmd-git on AUR using pamac.

Before executing simple example:
[image]

During/after executing simple example:
[image]

If I, during the execution of the simple script, move the headset the positions in the example are changing, so it recognizes the input, but neither the PSVR-displays nor my monitor changes/moves.

If anyone has ever experienced this or if any other input is required from me then please let me know.

TheOnlyJoey, 2021-01-24:
This is completely expected. There should not be a difference between the GL and Simple example (both are just simple applications using the OpenHMD library), It could be that one closes the driver successfully where the other gets interrupted?
In any case the warping is expected and actually what you want! But i will double check the applications to make sure they close down correctly/

Keep in mind the Simple example does not create any window, its just a terminal only application.
OpenHMD also does not do any window management at all (see https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode for more information).

closed 2021-01-24 by TheOnlyJoey

odecif, 2021-02-03:
Ok great, thank you for the clarification!