OpenHMD/OpenHMD #321: Image is not output to HTC Vive

happysmash27, 2021-01-24:
openhmd_opengl_example is able to correctly track my HTC Vive's gyroscope, but it does not seem to be able to actually output to the Vive, instead making a new window on my normal monitors. SteamVR can output to it fine, however. Is this a bug in OpenHMD, or is there some configuration I must do to make it work?

TheOnlyJoey, 2021-01-24:
We get this comment quite a lot, OpenHMD does not do any window/screen management. information can be found on https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode

closed 2021-01-24 by TheOnlyJoey

happysmash27, 2021-02-19:
Thank you so much! I had been looking for a way to enable extended mode for a very long time but was not able to until you linked me to this documentation. This allowed me to not only run OpenHMD, but also view my own stereoscopic images and watch 3D movies in Bino. 

I should have probably commented this earlier, when I was actually doing it, but better late than never.