OpenHMD/OpenHMD #323: Display off on HP Reverb G2

hi7, 2021-03-14:
I have build from master and when I run the simple example I got:
[EE] Failed to issue command 08: 17 00 01
[EE] Could not read config from the firmware
resolution:              0 x 0

l33tlinuxh4x0r, 2021-03-17:
I am thinking about getting this headset... Is it supported at all yet? Does it work at all with OpneHMD/Linux?

letoram, 2021-03-19:
Not supported yet, the magic sequence for turning the display on has changed from the G1 to the G2.

l33tlinuxh4x0r, 2021-03-21:
How difficult would it be to find the new magic sequence and add it to OpenHMD?

thaytan, 2021-03-22:
https://github.com/thaytan/OpenHMD/tree/dev-reverb-g2 lights up the reverb G2 for me if anyone wants to try that. There's lots more to do besides of course - eye rotations is wrong and there's very noticeable gyro drift on mine.

I didn't find a fix for the ` [EE] Failed to issue command 08: 17 00 01` error. It happens reliably on the first attempt after plugging in, but it works on the 2nd attempt.

hi7, 2021-03-22:
With https://github.com/thaytan/OpenHMD/tree/dev-reverb-g2 I got rid of the error message 
"[EE] Failed to issue command 08: 17 00 01" 
at second attempt to plug in, but I still got no light. Do I miss something.

thaytan, 2021-03-22:
Others on IRC have also said this doesn't work for them... but it does for me, so there's something else going on.

l33tlinuxh4x0r, 2021-03-22:
I haven't tried it yet but maybe the initialization code might be generated basted on HWID or serial number in addition so that it would work for you but not everyone else? Just a theory. I'll boot to Linux and try in a bit.

thaytan, 2021-03-22:
There is nothing uniquely extracted from my headset in that sequence - I just fiddled with the timing of the G1 sequence and added some retries to the config retrieval

l33tlinuxh4x0r, 2021-03-22:
OK so I tested it... My display doesn't light up also I needed to run as root to get it working. The simple test program then seemed to work. I will try running a second time to see if it wakes up. 2nd time it still doesn't light up but it is "working".

thaytan, 2021-03-22:
If you need to run as root, then you are missing a udev rule. https://gitlab.freedesktop.org/monado/utilities/xr-hardware has a rule for the G2 already.

l33tlinuxh4x0r, 2021-03-22:
> If you need to run as root, then you are missing a udev rule. https://gitlab.freedesktop.org/monado/utilities/xr-hardware has a rule for the G2 already.

Thanks.

l33tlinuxh4x0r, 2021-03-22:
I get the tracking now but no screen on.

thaytan, 2021-03-22:
I guess when we understand why it's working for me it'll be obvious in hindsight.

l33tlinuxh4x0r, 2021-03-22:
> 
> 
> I guess when we understand why it's working for me it'll be obvious in hindsight.

What OS are you using?

l33tlinuxh4x0r, 2021-03-22:
> 
> 
> I'm testing on a Fedora system

Have you tried using a Live system or a fresh install on another disk to see if you can replicate your results?

l33tlinuxh4x0r, 2021-03-22:
OK... I got the screen to come on!!! I'm using Archlinux with X11, gnome and the latest branch from thaytan.

This time I did ```git clone --branch dev-reverb-g2 https://github.com/thaytan/OpenHMD.git
```
instead of just git clone. Also it didn't seem to work the first time. Now it is lit up... It doesn't want to turn off though.

thaytan, 2021-03-22:
Ah. If you didn't switch branches before, it would not have been running the Reverb init code, and just treating it as an unknown WMR headset.

l33tlinuxh4x0r, 2021-03-23:
> 
> 
> Ah. If you didn't switch branches before, it would not have been running the Reverb init code, and just treating it as an unknown WMR headset.

Yeah that was the issue... I'm curious if that is why other people were having issues as well.

l33tlinuxh4x0r, 2021-03-24:
I am getting my headset to turn on 100% of the time. I even tried to play steamvr using SteamVR-OpenHMD however I couldn't get that to work.. It did wake up the headset but it wouldn't let me display anything... It just displays as another desktop. I can't use it for VR.

thaytan, 2021-03-25:
This is just init code - the driver still needs to report the right screen resolution and viewport mappings before SteamVR will be able to find it, and distortion info if we want it to look any good

hi7, 2021-03-25:
I'm on Fedora too and sometimes the G2 got detected with some flicker on my main monitor but no light on the G2 with dev-reverb-g2 branch.

thaytan, 2021-03-25:
> I'm on Fedora too and sometimes the G2 got detected with some flicker on my main monitor but no light on the G2 with dev-reverb-g2 branch.

Are you able to test on Windows on the same hardware / ports?

hi7, 2021-03-25:
SteamVR/WMR works with the G2 on Win10, if you mean that.

thaytan, 2021-03-25:
Thanks yes - there's enough people have trouble with G2 even under Windows that I want to make sure it's expected to work when testing here :)

l33tlinuxh4x0r, 2021-03-26:
> 
> 
> This is just init code - the driver still needs to report the right screen resolution and viewport mappings before SteamVR will be able to find it, and distortion info if we want it to look any good

Is anyone working on this? If not could you please point me in the right direction so I can try to tinker with it.

thaytan, 2021-03-27:
I probably won't be pushing it further in the next few weeks.  https://github.com/thaytan/OpenHMD/blob/670a2e666d217a4f218239c3e96f8f3e709ab868/src/drv_wmr/wmr.c#L533 is the right area if you want to play. I think it should be getting the right resolution details from the config blob already actually - so I'm not sure why Steam didn't find it.

l33tlinuxh4x0r, 2021-03-28:
Just rebuilt everything as root and installed the latest OpenHMD thaytan version and it is working in steamVR! only problem is that the video is rotated 90 degreees to the right. So I need to flip the image 90 degrees left to be able to test more. P.S. I had to enable direct display mode otherwise it just overlays stuff on my desktop. With direct display mode it appears to be working!

l33tlinuxh4x0r, 2021-03-28:
Looks like the controllers aren't working either and no VR programs will run but the headset is active and at the "mountain" screen 90 degrees flipped.

l33tlinuxh4x0r, 2021-03-28:
I got no man's sky to run, no controllers and everything is 90 degrees rotated... hope that this helps. Also not as clear as in windows so maybe the lens distortion is wrong.
[image]

l33tlinuxh4x0r, 2021-03-28:
For anyone that wants to test/help, I'm posting how I got it _working_.

```
mkdir VRStuff
cd VRStuff
git clone --recursive https://github.com/ChristophHaag/SteamVR-OpenHMD.git
cd SteamVR-OpenHMD/subprojects
rm openhmd -R
git clone --branch dev-reverb-g2 https://github.com/thaytan/OpenHMD.git openhmd
cd ..
mkdir build
cd build
cmake ..
make
cd ../..
git clone --branch dev-reverb-g2 https://github.com/thaytan/OpenHMD.git
cd OpenHMD
mkdir build
cd build
cmake ..
make
sudo make install
cd ../..
cd SteamVR-OpenHMD
./register.sh
```
That should get you up and running... Don't forget to enable direct display mode once you get SteamVR working so that it displays on your headset.

fluffrabbit, 2021-03-28:
You can install OpenHMD globally, but AFAIK you don't have to. SteamVR uses the copy in the SteamVR-OpenHMD folder. Correct me if I'm wrong.

l33tlinuxh4x0r, 2021-03-28:
> You can install OpenHMD globally, but AFAIK you don't have to. SteamVR uses the copy in the SteamVR-OpenHMD folder. Correct me if I'm wrong.

I will test this... It didn't seem to work before but I will test again and report back.

l33tlinuxh4x0r, 2021-03-28:
> You can install OpenHMD globally, but AFAIK you don't have to. SteamVR uses the copy in the SteamVR-OpenHMD folder. Correct me if I'm wrong.

You are correct... No need for a global install. Will do a reboot to be 100% sure in a bit.

Updated instructions:

```
mkdir VRStuff
cd VRStuff
git clone --recursive https://github.com/ChristophHaag/SteamVR-OpenHMD.git
cd SteamVR-OpenHMD/subprojects
rm openhmd -R
git clone --branch dev-reverb-g2 https://github.com/thaytan/OpenHMD.git openhmd
cd ..
mkdir build
cd build
cmake ..
make
cd..
./register.sh
```

l33tlinuxh4x0r, 2021-03-28:
I ran steam from terminal and got a bunch of ```[EE] unknown message type: ``` errors. The number changed every time I either moved the HMD error 23 iirc and a bunch of different error numbers when I moved the controllers.

fluffrabbit, 2021-03-28:
The `[EE]` errors are printed by OpenHMD AFAIK. I have not done this myself, but if things are generally working then I would blame it on deficiencies in OpenHMD's driver for your device.

l33tlinuxh4x0r, 2021-03-29:
I have made some changes to thaytan's code... First Headsets should always work now even on cold boot! No more having to run OpenHMD 2 or 3 times to get your headset to work. I also separated reverb and reverb g2 info and error messages to be more clear. The above problems are still a thing though... 90 degree rotation, controllers don't work etc.

I have updated instructions for anyone that wants to test. Also I put it in master not dev branch so it is easier to checkout.

```
mkdir VRStuff
cd VRStuff
git clone --recursive https://github.com/ChristophHaag/SteamVR-OpenHMD.git
cd SteamVR-OpenHMD/subprojects
rm openhmd -R
git clone https://github.com/l33tlinuxh4x0r/OpenHMD.git openhmd
cd ..
mkdir build
cd build
cmake ..
make
cd ..
./register.sh
```

Here is my repo for review https://github.com/l33tlinuxh4x0r/OpenHMD.git

thaytan, 2021-03-29:
That's a bit ugly. By rebasing on my repo's git master instead of upstream you've pulled in extra commits that can never got upstream (me updating the README to provide Rift status updates for example), and by squashing all the existing G2 commits into one you've thrown away attribution and made it hard to see what you've changed.

Finally, a potentially infinite loop reading the config isn't upstreamable. It has to give up at some point.

l33tlinuxh4x0r, 2021-03-29:
> That's a bit ugly. By rebasing on my repo's git master instead of upstream you've pulled in extra commits that can never got upstream (me updating the README to provide Rift status updates for [...]

Thanks for the heads up... I will try to fix the git repos... I'm not very good at git unfortunately.

I figured that the infinite loop was a bad thing. Maybe I could make it loop like 10 times or something?

l33tlinuxh4x0r, 2021-03-29:
I also made a hard limit of 10 tries for getting the config from the headset. That should be enough. My headset seems to need 3 checks, not sure about others. I might play around with it more to see if there is a more elegant way of getting the config. Maybe just pinging the headset a couple of times before trying? Not sure how to code that though.

thaytan, 2021-03-29:
I'd like to restructure all that code to:
 * Add an enum for known WMR headset types
 * Add a const array that maps VID/PID and description strings for the aux device to WMR headset type, init/deinit funcs and a headset name string for debug / pretty printing.

Also, your idea to keep things on master in your repo isn't bad. Jump back on IRC if you need git help.

thaytan, 2021-03-29:
If we could it would be better to figure out what the ` [EE] Failed to issue command 08: 17 00 01` return code actually means while reading firmware and handle it - that would avoid needing to retry at all.

thaytan, 2021-03-29:
I've merged https://github.com/thaytan/OpenHMD/pull/35 from @ph5 into my dev-reverb-g2 branch, which fixes config reads properly here.

l33tlinuxh4x0r, 2021-03-29:
> I've merged [thaytan#35](https://github.com/thaytan/OpenHMD/pull/35) from @pH5 into my dev-reverb-g2 branch, which fixes config reads properly here.

I can confirm that this works.

l33tlinuxh4x0r, 2021-04-04:
@thaytan In my experience with the new error 23 fix the following lines of code aren't needed anymore. For me it gets the config on the first try every time now.

```
	// Try twice to read the config. It fails the first time on HP G2
	// after a fresh plug-in
	if (config == NULL)
		config = read_config(priv);
```

l33tlinuxh4x0r, 2021-04-14:
Should we mark this bug as fixed since the device turns on now and open another bug for the rotation/sensors/controllers not working?

TheOnlyJoey, 2021-04-14:
> Should we mark this bug as fixed since the device turns on now and open another bug for the rotation/sensors/controllers not working?

Once a PR has been merged we can close this topic, which is not the case at this moment.

l33tlinuxh4x0r, 2021-04-14:
Is anyone doing a pull request? What is holding this up, just lack of time?

TheOnlyJoey, 2021-04-14:
> Is anyone doing a pull request? What is holding this up, just lack of time?

Please don't spam the issue, things happen when they happen.

LoganDark, 2021-04-19:
l'm interested in using OpenHMD as a replacement for WMR on machines without appx support (so can't run the Windows Mixed Reality Portal app). l own a HP Reverb G2 as well. ls there anything l can do to help the process of adding support for this headset?

l33tlinuxh4x0r, 2021-06-03:
After playing with the driver for hours... I figured something out. I can get the headset to stay the correct orientation if I comment out the following line in the source code. line 181 `*(quatf*)out = priv->sensor_fusion.orient;` The problem with that is there is no tracking at all if I do this (which is to be expected) however it tells me that this headset is doing something funny with ROTATION_QUAT I'll look into this more and hopefully I'll get lucky and find a fix.

thaytan, 2021-06-03:
The problem is that this headset has the IMU rotated differently to the other WMR headsets that the driver was written for. The correct solution is to read the IMU orientation from the firmware block. In the meantime, you can change the order of array accesses in `vec3f_from_hololens_gyro()` and `vec3f_from_hololens_accel()`. Instead of 1/0/2 for X/Y/Z axis ordering, just change the indices to 0/1/2 for X/Y/Z.

l33tlinuxh4x0r, 2021-06-03:
I have narrowed this down to the accel causing the rotation(I think). I don't understand what you are saying should be changed to get this to work though.

thaytan, 2021-06-03:
I have pushed a commit to https://github.com/thaytan/OpenHMD/tree/dev-reverb-g2 that implements what I mean.

l33tlinuxh4x0r, 2021-06-03:
I tested and added a patch that fixes a bug in the looking up and down functions of the headset. I submitted a pull request. https://github.com/thaytan/OpenHMD/pull/39

muskwasis, 2021-07-01:
How could we add more resolutions and refresh rates? I get 2280x1440@90hz, but the device also supports 4320x2160@90hz and 4320x2160@60hz.

isopix, 2021-07-01:
> but the device also supports ***@***.*** and ***@***.***

I never played with this device.

***@***.*** is native, the problem could be Graphics Card or
DisplayPort cable.
Try upgrading firmware or drivers.

xrandr should repport supported resolutions.

spiderbit, 2021-09-25:
@thaytan did you only add this rotation patch for the G2 or also the G1 because I have here a G1 and its 90 rotated to the right.

Btw because controllers don't work, I think it's possible to get the steamvr to not expect a controller or are they only used for pressing buttons or something?

spiderbit, 2021-09-26:
Somehow all broke after tracking and displaying worked, I tried 100 things reinstalling uninstalling different parts look over the Steamvr-openhmd doku but something went wrong maybe somebody can look over my vrserver.txt and the output of vrstartup and tell me what I did wrong. Did thaytan g2 branch as subproject and usde the same commands but I get some "error occured 203 in steamvr"
[vrserver.txt](https://github.com/OpenHMD/OpenHMD/files/7230687/vrserver.txt)
[run.log](https://github.com/OpenHMD/OpenHMD/files/7230688/run.log)

spiderbit, 2021-09-27:
Well tried reinstall steam like 10 times the first time it worked after fresh installation for 1 time or so to set it up but then it did not work again.
Even modified the PKGBUILD for openhmd-steamvr package, but of course didn't help, but maybe can do something with it.
[PKGBUILD.txt](https://github.com/OpenHMD/OpenHMD/files/7232564/PKGBUILD.txt)

thaytan, 2021-09-27:
203 error is apperently VRInitError_Driver_NotLoaded, but I don't have any guesses as to exactly what is wrong, sorry.

[29 more replies]