OpenHMD/OpenHMD #330: Oculus Rift CV1 kills all usb ports after a short while

BruebachL, 2021-10-02:
Hey,
everything works fine for a while but after a short time (usually within 10-15 minutes), the rift seems to mess up and then proceeds to interfere with any and all usb ports, forcing a reboot.

I'm running Arch Linux (Hardened) on an MSI B350 PC Mate motherboard. The rift and its two sensors are connected to the USB 3.0 ports on the motherboard.

I think this is an OpenHMD issue and not a hardware fault because everything works just fine with this exact same setup on Windows with proprietary drivers.

Please let me know if there is any more information I can supply.

thaytan, 2021-10-16:
I don't think there's anything OpenHMD can do here - it's just a consumer of the USB interfaces the kernel provides.

I've seen lots of people have trouble with the AMD USB implementation on other headsets, so my guess is that there's something a little strange about the USB hardware. The Windows USB drivers might have a workaround in place that the Linux kernel doesn't that makes it work there.

BruebachL, 2021-12-27:
Some good soul somewhere fixed this for me. I suppose that workaround you speculated about got implemented. In any case, everything works as expected now.

closed 2021-12-27 by BruebachL

thaytan, 2021-12-28:
Thanks for the update!

Dynamicrhapsody, 2022-02-17:
Right, so this is still an issue. This has actually been an issue for me for quite a while unfortunately. Oculus has caused me lots of headaches, just from the HDMI not being 'detected', all my USBs rebooting, all of my monitors rebooting, sometimes not even turning back on, just turning off. Too many headaches, happens every time I plug in. I've debated getting a new headset and with as many issues as I get, it's bound to be worth it.

thaytan, 2022-02-17:
> Right, so this is still an issue. This has actually been an issue for me for quite a while unfortunately. Oculus has caused me lots of headaches, just from the HDMI not being 'detected', all my USBs rebooting, all of my monitors rebooting, sometimes not even turning back on, just turning off. Too many headaches, happens every time I plug in. I've debated getting a new headset and with as many issues as I get, it's bound to be worth it.
> 
> additional info - Oculus support hasn't done anything outside of telling me that I shouldn't use a USB extension (I don't) or that I should turn it (my CV1) off and on again. Yes, they seriously said that to me.

Are you talking about the Windows Oculus driver, or about OpenHMD?