OpenHMD/OpenHMD #331: Rift CV1 Not Tracking Position

Valkin-27, 2021-10-03:
Hello,

I tried setting up my Oculus Rift CV1 on OpenHMD-SteamVR on my Ubuntu system. And I can get steamVR to launch okay, but as soon as I am in VR, I am stuck in the ground and my controllers/headset can rotate where they are, but they cannot actually move. I have both my cameras plugged into a USB-C, 3.2 hub, but neither of their lights are on. In addition, the sample (simple) code in terminal will output rotational information, but all positions are just 0.

I followed Jan's guide: https://noraisin.net/diary/?page_id=1048 and have all the installed packages as said here. But still nothing works. I am hoping someone has experienced and fixed this at some point.

Thank you so much!

[image]

System information:
OS: Pop OS (Ubuntu 21.04)
32GB GDDR4-3600mHz ram
GPU: GTX 1060
CPU: Ryzen 3900X

concertedrxn, 2021-10-17:
I am not the OP, but I ran into the same issue. For me, there were two problems getting set up. The first problem was that the xr-hardware package does not include rules for the CV1 sensors, but I was able to fix the problem by creating my own udev rules. The second was that one of my three sensors was attached to a USB 3.1 gen 2 port. Even with the proper rules in place, it refuses to work unless I move it to a USB 3.0 (USB 3.1 gen 1) port. (Oddly, the CV1 HMD itself works fine on the USB 3.1 gen 2 port, but none of my three sensors do.)

Valkin-27, 2021-10-17:
Hi yes I am sorry, the issue was resolved. I accidentally didn't put the right OpenHMD repository in the SteamVR-OpenHMD folder and forgot to set this to closed.