OpenHMD/OpenHMD #333: Rift S - some news ?

gamersalpha, 2022-02-01:
Hello evrybody 

do u have some news for Rift S ?

i try last time (more than 6 month)

and the Headset looks working, but i got black screen on the headset 

can i try again now ?

thanks all

Nan123721, 2022-02-07:
I have been wanting this for a while too, I really appreciate the communities work on this, especially since the Rift S is a neglected piece of hardware, Can't wait!

wsippel, 2022-04-15:
I just noticed this blog post from last week: https://www.collabora.com/news-and-blog/blog/2022/04/05/visual-inertial-tracking-support-for-monado-openxr/

TLDR (as far as I gather): A Collabora intern implemented vsual-inertial tracking for Monado, enabling inside-out tracking for Windows Mixed Reality headsets. If I understand the basics and current state correctly, the Rift S works pretty much the same way WMR headsets work, and it's already easily possible to read the video feeds coming from the Rift S, so would it be possible to adapt Collabora's work for the Rift S?

thaytan, 2022-05-16:
I have a WIP branch with the Rift-S driver ported to Monado at https://gitlab.freedesktop.org/thaytan/monado/-/commits/dev-rift-s

gamersalpha, 2022-07-30:
hello all, can u have some news ?

what works and not works ?

i want try again thanks all :)

thaytan, 2022-07-31:
All current development is happening in Monado, not in OpenHMD. The latest status is here: https://youtu.be/Bgl8xrk6BS8 but it isn't easily usable for other people yet. Jump on the Monado discord if you want to follow my development.

gamersalpha, 2023-01-03:
Hello there, how are u some news 

we are the 03/01/2023

some news of the project please

thaytan, 2023-01-06:
The Rift S driver was merged into Monado in September. It still needs controller tracking, but head and hand tracking work

gamersalpha, 2023-01-08:
Monado, i don't know this project

and i see no one using it on internet ???

thaytan, 2023-01-09:
https://gitlab.freedesktop.org/monado/monado

gamersalpha, 2023-01-09:
Hello thanks for the link

so i understand, that's means i can play fully with my rift S with steamVR ?

Controller OK
HeadTracking OK
Hand Trackin OK 

i just have to found how compil it on my linux 

thanks

thaytan, 2023-01-10:
Controllers are still only rotational tracking