OpenHMD/OpenHMD #336: Nolo CV1 Pro - where are id that need to be changed?

JacekJagosz, 2022-03-10:
I bought what I thought was a Nolo CV1 but it ended up being a CV1 Pro. As far as I know the only way it differs is the `idProduct` and `iProduct`.

Where are those referenced, so I could add ids of the Pro version and try if the driver works with it?
I know this will also need a change in `xr-hardware` and I [already created an issue there](https://gitlab.freedesktop.org/monado/utilities/xr-hardware/-/issues/8).
```
idVendor 0x28e9 GDMicroelectronics
idProduct 0x0302
iManufacturer  1 GigaDevice
iProduct  2 CV1_PRO_HEAD
iSerial 3 GD32F3XX-V1.0.0-4b5c6de
```
What changes are necessary to `nolo.c` then? I did change some things already but it is not enough yet for it to work:
https://github.com/JacekJagosz/OpenHMD/commit/44cd729734e1afd898ece996fc9db6a457cb4495

isopix, 2022-03-10:
Hi

It has different refresh rate for HMD, Nolo told it has better CPU.

I've got two sets of regular CV1 and wonder if they could be upgraded
to 'Pro somehow'.

Could you make firmware dumps?

There's instructions and needed software/tools:
https://github.com/DJPlaya/Tesseract-Cobra

cheers,
Krzysztof Krakowiak

JacekJagosz, 2022-03-10:
@isopix I tried, but I have problems with dumping firmware, described them here: https://github.com/DJPlaya/Tesseract-Cobra/issues/3 Maybe you could help?
From the good news, CV1 Pro just got added to `xr-hardware`: https://gitlab.freedesktop.org/monado/utilities/xr-hardware/-/issues/8

JacekJagosz, 2022-03-10:
Also, here is the full output of `lsusb`:
```
 Device: ID 28e9:0302 GDMicroelectronics CV1_PRO_HEAD
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x28e9 GDMicroelectronics
  idProduct          0x0302
  bcdDevice            1.00
  iManufacturer           1 GigaDevice
  iProduct                2 CV1_PRO_HEAD
  iSerial                 3 GD32F3XX-V1.0.0-4b5c6de
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0069
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
          Report Descriptor: (length is 33)
            Item(Global): Usage Page, data= [ 0x8c ] 140
                            Bar Code Scanner Page (POS)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Local ): Usage, data= [ 0x03 ] 3
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x00 0xff ] 65280
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x00 0xff ] 65280
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
[... 79 lines cut ...]
          wDescriptorLength      33
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
cannot read device status, Resource temporarily unavailable (11)
```

JacekJagosz, 2022-03-13:
I did some progress! Now I am getting some output from the first device! You can see my progress here: https://github.com/JacekJagosz/OpenHMD/commits/master
I am completely new to this, would really appreciate any tips!
I have 3 observations for further work:
1. I still am only getting data for device 0 (Class HMD), but only zeroes for controllers. The Pro is seen as 3 different hid devices, and as far as I know base CV1 was just one. And my tweak is not enough.
```
[   63.117770] usb 1-1: new full-speed USB device number 3 using xhci_hcd
[   63.283116] usb 1-1: New USB device found, idVendor=28e9, idProduct=0302, bcdDevice= 1.00
[   63.283130] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   63.283133] usb 1-1: Product: CV1_PRO_HEAD
[   63.283136] usb 1-1: Manufacturer: GigaDevice
[   63.283139] usb 1-1: SerialNumber: GD32F3XX-V1.0.0-4b5c6de
[   63.310392] hid-generic 0003:28E9:0302.0008: hiddev3,hidraw7: USB HID v1.11 Device [GigaDevice CV1_PRO_HEAD] on usb-0000:04:00.3-1/input0
[   63.321635] hid-generic 0003:28E9:0302.0009: hiddev4,hidraw8: USB HID v1.11 Device [GigaDevice CV1_PRO_HEAD] on usb-0000:04:00.3-1/input1
[   63.332394] hid-generic 0003:28E9:0302.000A: hiddev5,hidraw9: USB HID v1.11 Device [GigaDevice CV1_PRO_HEAD] on usb-0000:04:00.3-1/input2
```
2. I get `[EE] invalid type given to getf (18d)` which might point me what else needs to be fixed.
3. The data I am getting right now seems to be random, without any pattern to it. I guess it is because data from the pro must be different.
Here is my output of the simple example, if anyone was wondering:
```
OpenHMD version: 0.3.0
num devices: 7

device 0
  vendor:  LYRobotix
  product: NOLO CV1 Pro
  path:    1-1:1.0
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 1
  vendor:  LYRobotix
  product: NOLO CV1: Controller 0
  path:    1-1:1.1
  class:   Controller
  flags:   16
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

device 2
  vendor:  LYRobotix
  product: NOLO CV1: Controller 1
  path:    1-1:1.2
  class:   Controller
  flags:   0e
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

[... 86 lines cut ...]

rotation quat:           -0.282611 -0.172046 -0.338000 0.881071 
position vec:            -0.225400 2.576800 -1.717400 

rotation quat:           -0.181622 -0.150075 -0.336027 0.911908 
position vec:            -0.938000 2.488800 -3.168100 

rotation quat:           -0.248822 -0.107058 -0.243638 0.931272 
position vec:            0.426800 -0.110600 -2.372500 

rotation quat:           -0.343478 -0.156179 -0.306439 0.873914 
position vec:            1.584500 -1.460000 -2.063100 

rotation quat:           -0.473666 0.049761 -0.308525 0.823393 
position vec:            2.869700 -2.469700 1.626600 

rotation quat:           -0.435521 -0.020112 -0.322156 0.840317 
position vec:            0.261800 -1.792700 -1.776300 

rotation quat:           -0.347164 -0.010034 -0.351704 0.869299 
position vec:            2.970100 -2.798900 1.550800 

rotation quat:           -0.238371 -0.109497 -0.419012 0.869263 
position vec:            0.397800 3.111100 1.937200
```