OpenHMD/OpenHMD #339: Oculus Rift - expected behavior - example window but glasses dark?

MarcWeber, 2022-04-29:
I get a x window (using wmii). But the glasses stay dark. If I move the headset the view changes (rotation only).
xrandr shows only main screen of laptop as connected.

What is expected behavior ?

```
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Oculus VR, Inc.
  product: Rift (CV1)
  path:    0001:0022:00
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
resolution:              2160 x 1200
hsize:                   0.119340 
vsize:                   0.066300 
lens separation:         0.054000 
lens vcenter:            0.030840 
left eye fov:            1.379104 
right eye fov:           1.379104 
left eye aspect:         0.900000 
right eye aspect:        0.900000 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:                

rotation quat:           0.000012 0.000045 -0.000010 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.000011 0.000062 -0.000032 1.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104689 0.000056 -0.015244 0.994388 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104698 0.000074 -0.015239 0.994387 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.104701 0.000092 -0.015232 0.994387 
position vec:            0.000000 0.000000 0.000000 

```

thaytan, 2022-05-01:
The OpenHMD examples don't do anything to connect to Direct Mode / non-desktop outputs, so it just pops up a regular X window. To get output to the headset that way, you need to turn off the non-desktop flag so it appears as a normal monitor and configure it in the desktop Display settings. Better though is to run something that does do direct output - like Monado, SteamVR etc.

Also, for CV1 you probably want to use my branches with positional tracking development - see https://noraisin.net/diary/?page_id=1048

MarcWeber, 2022-05-06:
What I want is displaying X on the glasses. Eg be outside while sun is shining and be able to see something. About the no-desktop which belongs to xrandr eg see https://www.reddit.com/r/openhmd/comments/de2blx/openhmd_works_but_i_cant_get_the_display_to_work/ I will try and report. But I guess the samples or the README should give hints. Eg to view on your glasses see http://github.com/README#nodesktop or such.

thaytan, 2022-05-07:
https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode is the page to find

MarcWeber, 2022-05-07:
Is Xarcan the way to go for trying X11 with glasses ? Your guide worked.

closed 2022-05-07 by MarcWeber