OpenHMD/OpenHMD #340: Oculus CV1 No Display

DaRacci, 2022-05-18:
I've been trying to get my oculus to work to a few days now to no success, Everything works, tracking, sensors, view window on steamvr, controllers etc however the display will not turn on. It doesn't appear in xrandr or anything but when SteamVR is launched i can see the display using `nvidia-xconfig --query-gpu-info`. I've installed the xr-hardware udev rules, and tested with a variaty of openhmd versions and branches but still with no results.
Is there any advice to get this working?

thaytan, 2022-05-18:
In `driver_openhmd.cpp` of SteamVR-OpenHMD, there's some lines that sleep to let the display connection establish. Maybe that sleep isn't long enough for your Nvidia driver?

```
        /* Sleep for 1 second while activating to let the display connect */
        std::this_thread::sleep_for( std::chrono::seconds(1) );
```

Try making that bigger.

DaRacci, 2022-05-18:
If this is the case should i be able to see a error or something in a log or should i just give it a shot?

thaytan, 2022-05-18:
If SteamVR can't detect a display, it'll just pop up a VR window instead of displaying on the headset - which sounds like what you're seeing?

DaRacci, 2022-05-18:
> If SteamVR can't detect a display, it'll just pop up a VR window instead of displaying on the headset - which sounds like what you're seeing?

No so im seeing that through activating like a view window or something in steamvr

DaRacci, 2022-05-18:
[image]
this is how im seeing it atm

DaRacci, 2022-05-18:
And this is my GPU Info as well as neofetch
[image]

```Graphics:
  Device-1: NVIDIA GA102 [GeForce RTX 3090] vendor: Gigabyte driver: nvidia
    v: 515.43.04 pcie: speed: 16 GT/s lanes: 16 ports: active: none
    off: DP-1, DP-2, DP-3, HDMI-A-1 empty: HDMI-A-2,HDMI-A-3 bus-ID: 0a:00.0
    chip-ID: 10de:2204
  Display: server: X.Org v: 21.1.3 driver: X: loaded: nvidia
    gpu: nvidia,nvidia-nvswitch display-ID: :0 screens: 1
  Screen-1: 0 s-res: 7680x1440 s-dpi: 108
  Monitor-1: DP-0 pos: primary,left res: 2560x1440 dpi: 109
    diag: 685mm (26.97")
  Monitor-2: DP-2 pos: primary,center res: 2560x1440 dpi: 107
    diag: 704mm (27.7")
  Monitor-3: DP-4 pos: right res: 2560x1440 dpi: 109 diag: 685mm (26.97")
  OpenGL: renderer: NVIDIA GeForce RTX 3090/PCIe/SSE2
    v: 4.6.0 NVIDIA 515.43.04 direct render: Yes```

DaRacci, 2022-05-18:
I've tried with sleep times of 2 up to 5 with no results

DaRacci, 2022-05-18:
I can now see the display in xrandr however i still have 0 display on it

thaytan, 2022-05-18:
If it's showing up in xrandr and the display configuration UI now, but you still can't get it to light up then I think you have a hardware problem

DaRacci, 2022-05-18:
I've tested it under my previous windows install which works perfectly fine

DaRacci, 2022-05-18:
If it helps at all this is my steamvr log
```vrcompositor.txt Wed May 18 2022 18:03:23.058359

Wed May 18 2022 18:03:23.058359 - //==============================================================================================
Wed May 18 2022 18:03:23.058377 - ================================================================================================
Wed May 18 2022 18:03:23.058385 - ================================================================================================
Wed May 18 2022 18:03:23.058395 - vrcompositor 1.21.12 startup with PID=74689, config=/home/racci/.local/share/Steam/config, runtime=/home/racci/.local/share/Steam/steamapps/common/SteamVR
Wed May 18 2022 18:03:23.058402 - VR compositor 1.21.12 (v1647034158) Mixed starting up
Wed May 18 2022 18:03:23.070981 - CIPCPipe::ConnectPipe(SteamVR_Namespace) attempting connect to steamvr:SteamVR_Namespace
Wed May 18 2022 18:03:23.071010 - CSharedResourceNamespaceClient::Init(): received namespace data 74569
Wed May 18 2022 18:03:23.071023 - CIPCPipe::CreatePipe(VR_CompositorPipe_74569) bound to steamvr:VR_CompositorPipe_74569
Wed May 18 2022 18:03:23.071425 - Set fd limit to: 524288
Wed May 18 2022 18:03:23.072002 - [Settings] Load Default Json Settings from /home/racci/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Wed May 18 2022 18:03:23.072035 - [Settings] Load Default Json Settings from /home/racci/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Wed May 18 2022 18:03:23.072062 - [Settings] Load Default Json Settings from /home/racci/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Wed May 18 2022 18:03:23.072088 - [Settings] Load Default Json Settings from /home/racci/Downloads/SteamVR-OpenHMD/build/resources/settings/default.vrsettings
Wed May 18 2022 18:03:23.072187 - [Settings] Load Default Json Settings from /home/racci/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Wed May 18 2022 18:03:23.072420 - [Settings] Load Json Settings from /home/racci/.local/share/Steam/config/steamvr.vrsettings
Wed May 18 2022 18:03:23.072899 - Events created
Wed May 18 2022 18:03:23.073261 - Found bad mirror window settings: 
Wed May 18 2022 18:03:23.073274 - Creating CHmdWindowSDL!
Wed May 18 2022 18:03:23.103992 - Instance layers available:
Wed May 18 2022 18:03:23.104016 -  - VK_LAYER_VALVE_steam_overlay_32, version 4202632
Wed May 18 2022 18:03:23.104024 -  - VK_LAYER_VALVE_steam_overlay_64, version 4202632
Wed May 18 2022 18:03:23.104031 -  - VK_LAYER_VALVE_steam_fossilize_32, version 4202632
Wed May 18 2022 18:03:23.104036 -  - VK_LAYER_VALVE_steam_fossilize_64, version 4202632
Wed May 18 2022 18:03:23.104042 -  - VK_LAYER_MANGOHUD_overlay, version 4202631
Wed May 18 2022 18:03:23.104048 -  - VK_LAYER_MANGOAPP_overlay, version 4202631
Wed May 18 2022 18:03:23.104053 -  - VK_LAYER_KHRONOS_validation, version 4206803
Wed May 18 2022 18:03:23.104058 -  - VK_LAYER_NV_optimus, version 4206797
Wed May 18 2022 18:03:23.106006 - Instance extensions available:
Wed May 18 2022 18:03:23.106025 -  - VK_KHR_device_group_creation, version 1
Wed May 18 2022 18:03:23.106032 -  - VK_KHR_display, version 23
Wed May 18 2022 18:03:23.106038 -  - VK_KHR_external_fence_capabilities, version 1
Wed May 18 2022 18:03:23.106055 -  - VK_KHR_external_memory_capabilities, version 1
Wed May 18 2022 18:03:23.106063 -  - VK_KHR_external_semaphore_capabilities, version 1
Wed May 18 2022 18:03:23.106068 -  - VK_KHR_get_display_properties2, version 1
Wed May 18 2022 18:03:23.106074 -  - VK_KHR_get_physical_device_properties2, version 2
Wed May 18 2022 18:03:23.106079 -  - VK_KHR_get_surface_capabilities2, version 1
Wed May 18 2022 18:03:23.106085 -  - VK_KHR_surface, version 25
Wed May 18 2022 18:03:23.106091 -  - VK_KHR_surface_protected_capabilities, version 1
Wed May 18 2022 18:03:23.106096 -  - VK_KHR_wayland_surface, version 6
Wed May 18 2022 18:03:23.106102 -  - VK_KHR_xcb_surface, version 6
Wed May 18 2022 18:03:23.106107 -  - VK_KHR_xlib_surface, version 6
Wed May 18 2022 18:03:23.106113 -  - VK_EXT_acquire_xlib_display, version 1
Wed May 18 2022 18:03:23.106118 -  - VK_EXT_debug_report, version 10
Wed May 18 2022 18:03:23.106124 -  - VK_EXT_debug_utils, version 2
Wed May 18 2022 18:03:23.106130 -  - VK_EXT_direct_mode_display, version 1
Wed May 18 2022 18:03:23.106136 -  - VK_EXT_display_surface_counter, version 1
Wed May 18 2022 18:03:23.106141 -  - VK_KHR_portability_enumeration, version 1
Wed May 18 2022 18:03:23.106148 - Direct mode features: present
Wed May 18 2022 18:03:23.106154 - Requesting 9 instance extensions:
Wed May 18 2022 18:03:23.106159 -  - VK_KHR_surface
Wed May 18 2022 18:03:23.106165 -  - VK_KHR_xcb_surface
Wed May 18 2022 18:03:23.106170 -  - VK_KHR_display
Wed May 18 2022 18:03:23.106176 -  - VK_EXT_direct_mode_display
Wed May 18 2022 18:03:23.106181 -  - VK_EXT_acquire_xlib_display
Wed May 18 2022 18:03:23.106187 -  - VK_EXT_display_surface_counter
Wed May 18 2022 18:03:23.106192 -  - VK_KHR_external_memory_capabilities
Wed May 18 2022 18:03:23.106198 -  - VK_KHR_external_semaphore_capabilities
Wed May 18 2022 18:03:23.106203 -  - VK_KHR_get_physical_device_properties2
Wed May 18 2022 18:03:23.108841 - HMD deviceUUID is 5023beabdeb6a6cf
Wed May 18 2022 18:03:23.108861 - Looking for direct display through RandR
Wed May 18 2022 18:03:23.108869 -  - Root 0x1ef
[... cut ...]
Wed May 18 2022 18:03:28.012392 - External connection from /home/racci/.local/share/Steam/steamapps/common/SteamVR/bin/vrwebhelper/linux64/vrwebhelper 75828
Wed May 18 2022 18:03:28.084685 - Processing message VRMsg_CompositorNewSharedImage from vrwebhelper_main (74724) took 0.0051 seconds
Wed May 18 2022 18:03:30.997375 - Clearing dashboard transform because timed out
Wed May 18 2022 18:04:17.944125 - Clearing dashboard transform because interaction started on HMD
Wed May 18 2022 18:04:56.635893 - Clearing dashboard transform because interaction started on HMD
```

thaytan, 2022-05-18:
```
Wed May 18 2022 18:03:23.235113 - 1 modes on display:
Wed May 18 2022 18:03:23.235121 -  - 0: 2160x1200@89.998001Hz
Wed May 18 2022 18:03:23.235128 - Selected mode 0.
Wed May 18 2022 18:03:23.235499 - Direct mode surface: 0x55e0ac12db50
```

That bit looks like SteamVR is finding the display, and getting a Direct Mode lease. I don't know why it's not lighting up after that

DaRacci, 2022-05-18:
That's what i thought while looking over it, I'm honestly really lost because it works completely besides the display I can even see steamVR getting input from the sensors and waking up / unpausing im just confused

VDavid003, 2022-07-23:
I have the exact same problem. CV1, display is black with orange led. Tracking in steamvr and the opengl example works, and if I set extended mode in my xorg config (nvidia card), xrandr sees the display but I have no output.
I have a Windows dualboot and the official Oculus app works fine under Windows.
Could this be a HMD firmware update? Or something like a new nvidia driver breaking it?

VDavid003, 2022-07-23:
Figured it out! I downgraded my kernel to the latest 5.15 lts, and my nvidia drivers to 495.46 and my CV1 came to life!
I'm guessing this is more of an nvidia driver problem rather than a kernel problem.

thaytan, 2022-07-23:
Thanks for reporting back!

DaRacci, 2022-08-01:
> 

Im unable to test as i don't have my rift currently but if thats the solution jesus thats annoying

VDavid003, 2022-08-02:
I guess we could report this to nvidia somehow? But we'll need more testing before that to see if it's an nvidia problem to begin with.

DaRacci, 2022-08-02:
> I guess we could report this to nvidia somehow? But we'll need more testing before that to see if it's an nvidia problem to begin with.

If No one else is able to verify the issue I'll be able to test sometime middle of September as that's when I'll have access to my oculus again.

thaytan, 2022-08-12:
Someone else reported on reddit similar symptoms that were fixed by downgrading their Nvidia driver: https://www.reddit.com/r/openhmd/comments/wkhvr9/comment/iju9lw3/

midget3111, 2023-01-17:
Yep, also encountering this issue :c

DaRacci, 2023-01-18:
> Yep, also encountering this issue :c

Tis sad but the only fix is either hoping nvidia fix it eventually or downgrading

RetroZelda, 2023-02-25:
still happening with driver version 525.85.12

thaytan, 2023-02-25:
https://github.com/ValveSoftware/SteamVR-for-Linux/issues/520 seems related, and points to
https://github.com/ValveSoftware/SteamVR-for-Linux/issues/355#issuecomment-635919607 as a possible workaround

Maraca177, 2024-01-28:
Any news ? Oculus Rift CV 1 on linux finally ?? any magic combination ?

secretwolfie, 2024-12-24:
23rd December 2024 - Any Updates regarding this situation? Nvidia Driver 565.77 on Fedora 41.