OpenHMD/OpenHMD #342: Oculus CV1 IMU latency problem

VDavid003, 2022-07-23:
On the Oculus Rift CV1, I have some heavy "input lag"-like latency issues with the IMU.
On OpenHMD master branch (3DOF CV1 support), I only see that while turning my head in the hmd, the whole thing turns late, and seems to be choppy.
On Thaytan's branches with 6DOF CV1 support I see the following message being spammed when I open one of the demos:

`[WW] Exposure timestamp 2396847104 was more than 1.5 IMU samples earlier than IMU ts 2396857332 by 10228 µS`

It seems like the difference increases until it reaches around 19000 µS then it jumps down and climbs up again.
Am I the only one experiencing this? It is not a performance issue as it happens with the simple example too (no 3d rendering), and the cpu can easily handle it (i5-12400f)

thaytan, 2022-07-23:
That kind of message (`[WW] Exposure timestamp 2396847104 was more than 1.5 IMU samples earlier than IMU ts 2396857332 by 10228 µS`) implies that OpenHMD didn't get scheduled in time and missed USB traffic. Make sure you're building in release mode (`cd build; meson -Dbuildtype=release`) to get compiler optimisations and see if that helps?

In general I don't see the lag or choppiness.

`It seems like the difference increases until it reaches around 19000 µS then it jumps down and climbs up again.`

VDavid003, 2022-07-23:
I did build in release. but I noticed now that after unplugging the HMD from USB and plugging it back in, the latency got much better. It's been plugged in for a long time (probably months), since I don't really unplug it and on Windows it worked fine.

thaytan, 2022-07-23:
That's.. interesting. I can't think of any reason why that would happen. If it happens again it might be interesting to record a USB trace with wireshark

Maraca177, 2024-01-28:
VDavid003, hello guys, I just signed up to find out if you guys did finally find a way to make the oculus rift works under linux again, was it a matter of waiting for the kernel to be fixed drivers from nvidia ? mesa again ?? I am still trying to make this wonderfull glasses to start working again under linux, ubuntu Centos Arc Linux. but I am kind of lost if I still need to go back to the kernel and drivers...doesn´t make any sense, didn´t Nvidia fixed it ? Thankss it will be a pleasure to try my best as well to grow and learn in here so may be someday I can help others as well. Cheers and God Bless...