OpenHMD/OpenHMD #343: Rift CV1 working under Linux 22.04 - full offline install

CBtiger, 2022-07-29:
Not really and issue, but hopefully some shortcut for other Rift users in Linux. 

I got the CV1 working after ages of "black image, organge LED".  

Here my configuration
* Ubuntu 22.04 installation, older AMD CPU, nVidia 1050Ti graphics card, HMD connected directly to the HMD port. (Adpaters and extensions lead to black image)
* many nvidia-drivers do not work. Nouveau is missing vulkan. I had success with the Ubuntu repository packet "nvidia-driver-450". The default one for 22.04 (V515) did definetively no work.
* Installation OpenHMD compiled from source, installed with "sudo make install". With the repository packet it did not work
* Installation Monado compiled from source with all dependencies, installed with "sudo make install". The repository packet did not work as its missing the OpenHMD driver. 
* Installation OpenXR packages from the 22.04 repository, except "openxr-monado" - this again is missing the openHMD driver
* Making the symlink point to the self compiled monado-service driver (as per instructions)
Thats it. I did not install SteamVR because I had an offline PC and it would not install. Its not required. The CV1 only shows an image after starting the monado-service and when the xrandr non-desktop=0 (which should be default)
* fully offline copy of the standard Ubuntu 22.04 repository (with apt-mirror)

The experience Its not fully smooth yet, but usable. Can't comment on rendering quality, though
How run:
 1. start monado-service in a separate terminal. This only worked for me with these settings
   export XRT_COMPOSITOR_FORCE_XCB=1
   export XRT_COMPOSITOR_XCB_FULLSCREEN=1
   monado-start        # note: no "sudo"
2. open the nvidia-settings tool and rearrange the screen such, that the Monado fullscreen window is on the monitor "Rift". This is required each time the service is starting and the only annoying issue. 
3. Start an application, e.g. Blender, enable VR in the preferences and start the VR mode in the SidePanel. The user manual describes that quite well
done.

The frame rates are rather low, but I guess thats to expect with the hardware. However I am very pleased with the result. Btw, I do not have any controllers to test. 

To debug the whole chain bottom up works great. Thtas why Im posting this here: If OpenHMD does not get an image in to the HMD with its "opengl" demo, there is no need to even look at Monado or OpenXR-SDK. 
Monado is crashing or giving erros with the default settings and with other parameter settings. For compilation I did not use any cmake flags, but when not all dependencies are installed the required "XCB" support will be missing.  

If someone has an idea how to get rid of step 2 above  I would say CV1 in Linux could be really ready for day-to-day use.

thaytan, 2022-07-29:
If you're using an Xorg DE login, you should be able to use the headset in direct mode. That will improve your performance somewhat. I've had no trouble doing that on my machine with an nvidia card.

It's not clear if you're using the upstream OpenHMD CV1 driver, or my branch at https://github.com/thaytan/OpenHMD/tree/rift-kalman-filter 

If you use my branch, you should be able to get positional tracking.

CBtiger, 2022-07-29:
@thaytan, yes, I do use your kalman filter branch. I had not tried the camera yet, first priority was to get an image at all. 

Performance is not critical, I am getting about 10-20fps, so its far away from fluid anyway... 

You mean what I see in step 2 is the consequence of non working direct mode? I am not full sure what your mean with Xorg DE (DE as DesktopEnvironment?). Currently I am running xfce which should not add a lot of overhead. Hover I would like to try what you suggest, but I am not sure how to get there. Xorg is installed obviously, but I am not aware of an alternative login. In the Debian/Ubuntu repository I did not see what I would need to install. Can you please elaborate a bit more on that?

thaytan, 2022-07-30:
> You mean what I see in step 2 is the consequence of non working direct mode?

Yes. In direct mode, the HDMI output the headset is connected to is set to `non-desktop=1` mode, and Monado and SteamVR should find it at startup and use it directly, without the overhead of going through the X server and desktop compositor.

> I am not full sure what your mean with Xorg DE (DE as DesktopEnvironment?). Currently I am running xfce which should not add a lot of overhead. Hover I would like to try what you suggest, but I am [...]

I mean using Xorg vs Wayland as the graphics server. The wayland support for direct mode access is not widely available yet, so it's best to stick with Xorg for VR stuff for now. Usually it's an option in a dropdown somewhere at the login screen to choose what kind of desktop session you want to log into.

CBtiger, 2022-07-30:
OK I can confirm that I am running X11. 
```
 echo $XDG_SESSION_TYPE
   X11
```
Just for testing I Installed Weston to see the differences, but here monado-service fails with errors also when I try to ..._FORCE_WAYLAND=1. With what you say I think its no use to follow up on that road. 

I noted that already OpenHMD's opengl example fails when I set non-desktop=1. Is that different in your setup ?
In both cases on the desktop I see the 3D window with the distored views. The difference is that while the demo runs in nvidia-settings I see the "Oculus VR Inc. Rift" monitor disabled (no image in HMD) or enabled (image in HMD).

thaytan, 2022-07-30:
The OpenHMD opengl example does not do direct mode stuff, it will always pop up a window. You need to test with Monado or SteamVR for that - their compositors will try and find a direct mode output when they start.

CBtiger, 2022-07-30:
Right, thanks I was not aware of that distinction. 

When I run monado-service with non-desktop=1 do do not see that happening. I still see the window on the desktop and no image in the HMD. As on my system monado-service is giving errors when run without XRT_COMPOSITOR settings, I would be interested to know if your setup behaves the same.
 - monado-service started with FORCE_XCB=1 and XCB_FULLSCREEN=1
 - nvidia-settings show the Rift monitor as "disabled" when the service run.

thaytan, 2022-07-30:
`FORCE_XCB=1 and XCB_FULLSCREEN=1` - these env vars force Monado to create a plain X window for output, disabling direct mode.

Take a look at https://monado.freedesktop.org/direct-mode.html You might not yet have the `Option "AllowHMD" "yes"` setting for nvidia?

CBtiger, 2022-07-30:
Indeed I have the AllowHMD set to yes. I will try to remove it. 
But when I got you right, I only have a chance when I find a way to run monado without these two env vars. Does it run at your system with only default env vars? 
Thanks for your patience

thaytan, 2022-07-31:
On my nvidia system, I test with the nvidia driver (currently 510.68.02) and without `AllowHMD` or other monado env vars.

CBtiger, 2022-07-31:
Monado keeps crashing. I always get
` segfault at 07F7334b7c3d3 sp .. error 4 in libnvidia-glcore.so.510.73.05`

I installed the reository version of the 5.10 driver (contains 510.73 driver not 510.68 the one before). I re-compiled monado and confirmed that I did not get any errors. Settings were
```
 "AllowHMD" "No"
 non-desktop=1
```
Its basically the same I had with the V450 drivers, Driver 510.68 is only available as DEB download from nvidia, but I do not really expect a different result.

CBtiger, 2023-08-06:
It seems I never had a working direct mode. 

I did set up 2 similar PCs with the same GTX1050 TI card. They got the same software  - all compiled from source (in release mode): first OpenHMD (Kalmanfiler version), then Monado (August 2023 head version) and finally the current OpenXP SDK with hello_xr and xrgears demos for testing. One PC got nvidia drivers V510, the other V525. Both run Xubuntu 22.04 only in X11 (no wayland)

The CV1 is recognised by USB and all OpenHMD demos are working fine. So no udev rule issue. 
The Monado driver starts without errors in X11 and prints the correct GPU and list the Oculus CV1. I have no camera and no controllers attached, just the Headset. 
The HDMI-0 is set to "non-desktiop=1" and is shown as "disconnected". The driver does not have any "Allow" configured and I am not using any environment variables.

Both OpenHMD simple demo and Monado-service list the correct resolution of 2160x1200. 
When the monado service starts up the screen flickers twice as it would be reconfiguring for the HMD display, but the display remains dark. When the "hello_xr" or the "xrgear" demo is started, they connect to the monado service and there are loads of "skipped old frame" warnings. So it looks like active rendering is taking place. 

Everything seems running perfectly, just no image. The display never wakes up. The LED nevet went white. It stays yellow. 

I even suspected the display was damaged, but with XCB_FORCE=1 and "allow" I get a desktop image in the HMD. So the hardware seems working fine.
I checked dmseg log but there were no errors. What other logs would be worth investigating?

I checked the other posts with a dark CV1 display but there did not seem to be a common root cause. 
One hint was the xorg.conf. Ubuntu does not have a xorg.conf by default, only single "/usr/share/X11/xorg.conf.d/" files. I switched one of the PCs  to "/etc/X11/xorg.conf" which should override othe others, but did not note a difference. What HMD specific settings would need to go into that file?

closed 2023-08-08 by CBtiger