OpenHMD/OpenHMD #344: aiming for a new release / milestone

spiderbit, 2022-10-06:
This project seems pretty direction less, while on many repos / branches there are interesting things.

The last 0.3.0 release is > 3 Years old, wouldn't maybe giving some goals and or deadlines / plans of release change that, even if it would only be some 0.3.99.1 or 0.4.0 pretest release.

I mean there are huge steps not production ready but close like initial wmr controller support, pretty close hp reverb (g1+g2) + other headset support. Do they have to stay another 5 years in some 99% finished forked repository?

Wallbraker, 2022-10-08:
You are now a owner :)

Wallbraker, 2022-10-08:
@thaytan you are now a owner.

thaytan, 2022-10-09:
> @thaytan you are now a owner. smile

The monkey's paw curls

FlooferLand, 2022-10-21:
I really like the idea behind this project, as a lot of VR headsets either have extremely sketchy telemetry-collecting software/drivers, simply do not support any platform besides Windows, or straight up have buggy abandoned software/drivers. And in a lot of cases it's at least 2 of those.

Apparently the owner of the project hasn't made a single contribution anywhere in over like 2 years.
Worst case scenario, someone could potentially fork this project? Or perhaps find some way to contact the owner?

I'm not sure if there are already forks working on continuing this project as i don't know how to look trough project forks more efficiently than manually clicking trough them, but most of the forks I've managed to find are tens of commits behind the main one.

Wallbraker, 2022-11-15:
Most of the former members now contribute to [Monado](https://monado.freedesktop.org/), which share the same goals (and more).

spiderbit, 2022-11-15:
Maybe but it's no real dropin replacement it seems, it would be great when tools would stop having steam as dependency :)

Wallbraker, 2022-11-18:
Monado doesn't have steam as a dependency, it can be built as a steam driver. But it works completely on it's own as well.

spiderbit, 2022-11-18:
I don't think Openhmd has a steam dependency. My issue is more on the application side I mean I even wrote it with the tools all the tools and games in steam, how much are there not for steam even tools you don't install over steam like "vr-video-player" don't work without steamvr.

The only thing that monado is used for it seems is some game engine and some demos. As long as nothing is build on top of it it's just a useless library.

Or have a rock solid steamvr plugin that works in 99.9999% of cases not the same wonky crap that comes with openHMD.

spiderbit, 2022-11-18:
well just tested monado on my arch machine with my vr reverb g1, got hello_xr more or less show me the cubes even they were not on the same position at all on the 2 sides, but then steamvr added the driver but nothing multiple tests...

As far as I can tell it tries always to open the steamvr app in a window instead of the HMD, and yes I know I should report that to the monado bugtracker :) but deal with that for at this point probably into years and tried to find help multiple times and multiple hardware and software combinations.