OpenHMD/OpenHMD #346: VRTEK WVR3 Driver (Rotate Screens – MOD 90º FOV @ 135º FOV)

AnonymousVR-VR, 2022-10-17:
Hello, good afternoon, I am contacting you, to ask you for a little help, I have seen your work on the “OpenHMD” website.

I am making a modification to a VR viewer the WVR3 VRTEK, the modification is simply to adjust the angles of the screens to have a much larger FOV, from the 90º degrees of FOV I can now have 135º FOV, you can follow the MOD in the Realovirtual website here:

https://www.realovirtual.com/foro/topic/51245/mod-visor-vr-los-90o-fov-135o-fov-anonymous-vr

The issue is that I have no way to configure the angles of the screens in the configuration file of the driver "default.vrsettings" in steamVR, I do not know what is the command or line that must be added to be able to tell OpenVR to change the screen angles.

Unfortunately I don't know how to create and compile new drivers with the OpenVR API.

I need help to get the screens to rotate at my choice, I ask you for help in this MOD, you who know how to program.

Thank you very much for your attention.
https://github.com/OpenHMD/OpenHMD/tree/master/src/drv_vrtek

thaytan, 2022-10-21:
You can modify the configuration of your screens by changing the projection and possibly modelview matrices in the driver

AnonymousVR-VR, 2022-10-21:
> You can modify the configuration of your screens by changing the projection and possibly modelview matrices in the driver

Unfortunately I don't know how to create and compile new drivers with the OpenVR API.

I need help to get the screens to rotate at my choice, I ask you for help in this MOD, you who know how to program.

I hope you can help rotate the screens, we would be very grateful in the Realovirtual ROV community.
Thank you very much for your attention.

AnonymousVR-VR, 2022-11-24:
> You can modify the configuration of your screens by changing the projection and possibly modelview matrices in the driver

You were on the right track, I have used the Openvr API, and I have managed with the ProjectionRaw function to create 2 individual functions for each eye and from there configure my screens.

virtual void GetProjectionRaw( EVREye eEye, float *pfLeft, float *pfRight, float *pfTop, float *pfBottom )
if (eEye == Eye_Left) {} else {}}

I have not been able to use OpenHMD on windows, I was not able to create the .dll drivers for SteamVR

Now I'm trying to create a universal application to change the parameters of GetProjectRaw for all viewers, this way, all the people who want to do the MOD with their viewer don't have to worry about compiling new drivers, besides it's very very difficult to make a 6DOF controller with all its functions in a homemade way.

For now I have only managed to make a very very simple utility, which from windows reads the ProjectionRaw parameters like this:

VRSystem->GetProjectionRaw(vr::Eye_Left, &left, &right, &top, &bottom);

I can read the parameters like this:
printf("Left Eye: %8.5f, %8.5f, %8.5f, %8.5f\n", left, right, top, bottom);

But I don't know how to change the ProjectionRaw parameters from an application that runs from the console.

You can help?

Thank you very much for your attention.