OpenHMD/OpenHMD #356: CV1 Feature: Display On Override

CBtiger, 2023-08-08:
This info is from the Monado website

> Because the HMD displays still fundamentally work as monitors and are only hidden via operating system/drivers, software that is not aware of the VR HMDs may still accidentally use the display. For example UEFI/BIOS and bootloaders may accidentally display on the HMD and not on a monitor.
> As a countermeasure some HMDs physically power off the display by default and require VR software **to enabled the display with an USB command (e.g. Oculus Rift CV1)**. With such an USB command the display will have to be physically powered on, before it is ready to be used for either extended or direct mode.

I got my CV1 in direct mode in Linux xcfe working, but only with the very old nvidia driver 390. All 5xx drivers seem to work ok and the OpenXR application seem to render to a hidden screen, but the CV1 display remains dark. This display On feature seems to be a main source of frustration also for other posters with CV1.

If really only that **USB ON command** is the problem, shouldn't the CV1 OpenHMD driver be able to override the driver and send the USB ON to the CV1 based on a user input? Could be for example a file in `~/.local/share/OpenHMD/overrideDisplayOn` which controls the on/off when existing, e.g.  "1"  switches the CV1 on and "0" off.  

I would try a test implementation but since I know nothing about the driver or how the Rift USB works, I would appreciate a tip where in the code the USB On signal is generated and which function would be a good place to add such a file check without impacting performance too much. Or maybe its a bad idea?

ChristophHaag, 2023-08-08:
I haven't tested it but the code in this repo should do tje trick, just ignore all the setup instructions beyond compiling and running it.
https://github.com/OhioIon/riftDriverPi

CBtiger, 2023-08-09:
Great, just what I was looking for.  Thanks.
I saw that the driver command could even enable Display, Audio and LED separately. 

I does cause the LED to turn white, but it seems a bit strange.
 
When the monado-service is running it can not access the USB as its blocked by moando. When the monado service is off, it does print out the correct HMD info but the LED does not turn white. I would have expected I could see it turn white after the command 0x1d is sent with byte3=0x01 but it does not. 

However, after running that little app once, the monado-service behaves differently after that. The screen flickers and the LED white. Even when no OpenXR application is running. It also works when the HMD is disconnected and re-connected again. That behaviour even survives a reboot. How can that be?

On the PC with the 510 driver when I start xrgears the following happens
* The LED turns yellow
* the screens flicker 
* monado service crashes (segfault)
*  xrgears showes the following errors
```
[ipc_receive_fds] recvmsg failed with error: no data!
XR_ERROR_RUNTIME_FAILURE in xrtCreateSwapchain: fiale to create swapchain
[e] xr.c55 | Failed to create depth swapchain 960354816! [XR_ERROR_RUNTIME_FAILURE]
```
When I disconnect the CV1 and run monado-service and xrgears it puts a window on the desktop and runs the xrgears fine. No segfaults.

CBtiger, 2023-08-09:
Its a theory only, but I recall that the Rift drivers displayed a certain "bit" to allow 3rd party drivers. It would have been part of the hardware and was cleared by default. Without this enabled things like SteamVR would not work. 

Since I only used very early dev drivers which could not set this bit, the CV1 hardware probably had this still set. That would explain that the CV1 would never light up when Monado activated the runtime.  If the small tool just set this "bit" as a side effect the CV1 would remember it and light up the display each time now. 
Does that make sense?