OpenHMD/OpenHMD #362: OCULUS RIFT STEAM VR XPLANE 11 OR XPLANE 12 AND LINUX, WHAT IS THE BEST COMBINATION DISTRO ETC... THANKS

Maraca177, 2024-01-28:
Well, First of all, Nice to post my first Title in here, normally I am the kind of guy who reads and try to find the solution on the problems others had before since I do use humble and legacy hardware normally, but in this case, I feel like I trully need to share my madness or happiness to have you guys all in here to fight again this kind of issues that makes us trying to feel even sorry for not being millionaire, since it is always a matter of buying new graphic card more ram etc...  the thing is that, believe it or not 5 years ago I made my Oculus Rift to work with a PHENON II AMD 20 years old already computer by viapassing the SSE limit by purpose to force us to always have to think on taking the easy solution if you have the money to always go for a better piece of .......... hardware. 

I started using the geforce gt 1030 and enjoyed the google earth VR and even XPLANE on it. until META changed the driver and since it is a new one the put also another limit for not using the GT 1030 AND IT WORKED REALLY NICE UNTIL THEY FORCED US TO STOP USING IT. now..... after readying what is happening in Linux also..... with the Nvidia drivers and kernel. I am really pissed off.

Any can tell me in here the details of the software using nowadays for making the Oculus Rift CV 1 WORK AGAIN PLEASE ? not sure if for not having to by pass the run with the script should I go to Arch linux or Centos, I would like to give it a try first on Ubuntu if possible, but not sure if I should use mesa, downgrade the kernel 6 down to other or drivers vulkan open gl or mesa ??? any help will be much apreciated. CONGRATULATIONS FOR THE TEAM WORK IN HERE YOU ARE GREAT GUYS !!!! GREAT WORK !!! ;) GOD BLESS

NewtSoup, 2024-01-28:
I think you are out of luck, my friend.

This project has not had any real work done on it in a long time.  Indeed the latest commit 2 weeks ago was to the readme.md saying that this project is currently unmaintained. Prior to that it's months or even years.

Until this project is picked up by someone you  might want to look at monado which has been worked on in the last year.

Maraca177, 2024-01-28:
Thanks for the answer NewSoup !! ;) I am reading and reading about monado and staff, I might give a try and test several kernels and versions, letś see. Thanks so much, do you use any VR ?

Maraca177, 2024-01-28:
[image]

Can´t believe now same thing now for a Chinese computer....have to by pass the SSE AGAIN ??? any suggestion ? to make it work ?

Maraca177, 2024-01-28:
a required extension for running the Open GL bridge was not supported by nvidia GT 1030 for running VULKAN ??? HAHAHAHA Cant believe it, I have been running with this GPU Vulkan for years, and now with new driver can´t ??? is it a joke ? :) lol