OpenHMD/OpenHMD #364: Cant get SteamVR to recoginize Rift CV1 but OpenHMD is working

OceanMan20, 2024-12-16:
I followed all the instructions from [here](https://gitlab.com/TacoCake/oculus-rift-openhmd-steamvr/-/blob/main/README.md). Using the test at the bottom that shows live position and rotational data, I can see that my headset is recognized. But when I start SteamVR it asks to plug in a headset.

I am on fedora 41 with KDE. I am pretty new to linux, I mostly stumbled my way here and after a few hours I was able to get OpenHMD working. Any help would be appreciated.

secretwolfie, 2024-12-24:
I'm having a similar issue, but even OpenHMD isn't working.

secretwolfie, 2024-12-24:
Update, plugging my device and reconnected and I'm getting responses through the test program, but the headset doesn't light up but the rotations are being detected

NewtSoup, 2024-12-24:
Please see the readme.md

https://github.com/OpenHMD/OpenHMD/blob/master/README.md

astoko, 2025-02-09:
Heyo, I have the same issue, did you guys figure it out?
My Sensor and headset is recognized via the openhmd_simple_example command to just test it and rotation is getting seen but i have no luck on getting the device recognized in steamvr.. 

**My System:**

[image]

**My Steam Client:**
```
Steam Beta Branch:  Stable Client
Steam Version:  1738026274
Steam Client Build Date:  Tue, Jan 28 2:50 AM UTC -08:00
Steam Web Build Date:  Tue, Jan 28 2:44 AM UTC -08:00
Steam API Version:  SteamClient021
```

**SteamVR System Report:**

```
SteamVR Version 2.9.6 (1738793859)
Steam:	Public (beta)
Tracking:	No Driver
OS:	Linux version 6.13.1-arch2-1 (linux@archlinux) (gcc (GCC) 14.2.1 20250128, GNU ld (GNU Binutils) 2.43.1) #1 SMP PREEMPT_DYNAMIC Fri, 07 Feb 2025 22:24:02 +0000 (0.0.0.0)
Direct Mode Graphics:	Unknown ()

System
Admin:	Yes
Async Reprojection:	
Performance drops:	0/1 0/2 0/3 0/4 0/0
User IPD (m):	Not yet implemented

Displays - Extended Display
```

If though im missing something please tell me, thanks in advance !!

OceanMan20, 2025-02-09:
> OS:	Linux version 6.13.1-arch2-1 ***@***.***) (gcc (GCC) 14.2.1 20250128, GNU ld (GNU Binutils) 2.43.1) #1 SMP PREEMPT_DYNAMIC Fri, 07 Feb 2025 22:24:02 +0000 (0.0.0.0)

I was not able to get it to work. Sorry, hopefully someone is able to get
it working and report back

thaytan, 2025-02-12:
These days I generally recommend that people use the OpenHMD driver via Envision: https://lvra.gitlab.io/docs/fossvr/envision/ which can help with getting the dependencies, compilation and setup right